bambulab/BambuStudio #4541: Bambu PETG-HF Profiles not shown nor detected

floppyxyz, 2024-07-24:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 & macOS Sonoma 14.5

### Printer

X1C

### How to reproduce

Loaded Filament in Printer on A2 & A4.
Printer showed detection and correct Filament types & Colors.
When Bambu Studio was opened while loading filament, the printer shortly showed the correct filament type, but then changes to "?".

The Profiles in Bambu Studio listed no PETG-HF, either in Printer Settings nor in Filament Selection on "Prepare view".

### Actual results

Windows:
[image]
[image]

MacOS:
[image]

Printer:
[image]

### Expected results

Correct detection and usage of petg-hf profile for printing

### Project file & Debug log uploads

[BambuDebugPETG_HF.3mf.zip](https://github.com/user-attachments/files/16360868/BambuDebugPETG_HF.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

invisibleninja06, 2024-07-24:
Same issue here, oddly enough there is the option to set it manually on the printer but it keeps wiping it once i go to set the next spool.

invisibleninja06, 2024-07-24:
Mine is not detecting either and if i try to set manually in either Bambu studio or on the printer directly it will show it set for a few seconds and than update to the question mark.

invisibleninja06, 2024-07-24:
oddly enough as a work around i was able to set the filament and have it stay by setting it using the bambu handy app.

azarius74, 2024-07-25:
Similar issue... 1) The AMS does not automatically detect it... it comes in as a "?".   I can manually change it to PETG-HF, but it is reverted to "?" when I try and sync in Bambu Studio.

I have the newest version of Bambu Studio - even uninstalled and reinstalled and there is no PETG-HF Profile anywhere...

kbzowski, 2024-07-25:
The same, PETG HF in AMS is not detected at all. 
I can print if I add in manually. 
BambuStudio 1.9.3.50

MennoBroxterman, 2024-07-25:
Look for a message somewhere in Bambu Studio which says, Configuration Update
. This will added PETG-HF to Bambu Studio.

[image]

TreZc0, 2024-07-26:
I am not receiving this message on Mac, is it not available there?

RotorWash, 2024-07-29:
ive updated firmware, clicked on the config update button in lower right corner but it still reverts to ? even after setting it to petg-hf manually.
Gotta be a bug.

wnsander232, 2024-08-03:
MacBook Pro 2018 running latest software, firmware, and app on iPhone 13.  PETG HF not listed as an option in config and will not auto detect.  Have reinstalled software and no change.   P1S w/ AMS.

kbzowski, 2024-08-04:
After profile update it works. It detected the green PETG, but the orange one in AMS does not want to detect. I had to add it manually. Maybe the tag is damaged.

discobean, 2024-08-06:
I have the same issue.  

Also when trying to resync it does not even detect the filament in A1 in Bambu Studio on Mac (even after update), instead it shifts the filaments in bambu studio less by 1.  So it thinks my external filament is A4, when it should be A5.

RotorWash, 2024-08-06:
manually assign petg basic to your petg-hf.
then go to prepare and sync the ams.
then in the same "prepare" menu, manually change the petg basic  to petg-hf.
that should make it stick. worked for me.
cheers

DanBao-Bambu, 2024-08-21:
The issue is due to Studio not having the Filament preset, which can be resolved by downloading the preset package.

wnsander232, 2024-08-21:
@DanBao-Bambu There is no option to download a preset package.  Can you post directions on how to do so?

ebeechsrc, 2024-08-21:
With Bambu Studio v01.09.03.50 on windows like others I can't use the PETG-HF.

On the printer itself, or the mobile app I can set the filament to PETG-HF, but if bambu studio connects, it sets the filament to a "?" and becomes unrecognised, even mid print using the filament.
Bambu Studio:
[image]

Mobile app:
[image]
[image]

I don't have an option within the desktop application to enable PETG-HF filament type, it just isn't there.
[image]

There doesn't appear to be a configuration for this filament in the current release in the [BBL/filament folder](https://github.com/bambulab/BambuStudio/tree/master/resources/profiles/BBL/filament) or a profile locally.
[image]