bambulab/BambuStudio #4543: Bambu Studio crashes unless I set CPU affinity

DroidSlicer, 2024-07-25:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home version 10.0.22631 Build 22631

### Additional system information

CPU: Intel 14th Gen i9-14900K
Memory: G. Skill 64GB DDR5-6000 (2x32GB)
GPU: Asus ProArt 4070Ti
Mobo: MSI Mag Z790 Tomahawk Wi-Fi (MS-7D91)

### Printer

Bambu Lab X1C

### How to reproduce

Load a complex/large file size model (or load provided Bambu Project).
Set printer to Bambu X1-Carbon.
Set filament profile to Bambu ABS (Stock).
Turn on Tree supports (auto).
Slice plate.

### Actual results

Bambu Studio will start slicing the model, get partially through the progress bar, and then the program completely closes on it's own. 

The only way I have been able to fix this issue is by setting the affinity every time I open Bambu Studio and only allowing it to use cores 0-7 (the 8 performance cores on this CPU).

### Expected results

The program should not crash. I am assuming it must have something to do with using too many cores on the CPU. This isn't a problem with other programs that I'm aware of. 

### Project file & Debug log uploads

[Bambu Studio Crashing Test.zip](https://github.com/user-attachments/files/16382860/Bambu.Studio.Crashing.Test.zip)
[Log Files.zip](https://github.com/user-attachments/files/16382861/Log.Files.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

frank-pian, 2024-07-26:
https://www.xda-developers.com/intel-cracks-the-cause-raptor-lake-instability/
It's an intel software bug that's been going on for a while now.

https://github.com/bambulab/BambuStudio/issues/3318
https://github.com/prusa3d/PrusaSlicer/issues/11845

DroidSlicer, 2024-07-26:
Ahh. Understood. Thanks for the quick response! Everything else about the program, I have loved so far! Thanks again!

closed 2024-07-29 by DroidSlicer