bambulab/BambuStudio #4544: Camera will not connect on PC app

LavonT07, 2024-07-25:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

14th Gen Intel I9 14900K

### Printer

Bambu Lab X1E

### How to reproduce

Send file to print, jumps to camera, will not connect, shows network issue but updates layers and percent complete. Bambu Handy also show live feed from camera

### Actual results

Send file to print, jumps to camera, will not connect, shows network issue but updates layers and percent complete. Bambu Handy also show live feed from camera

### Expected results

Send file to print, jumps to camera, will not connect, shows network issue but updates layers and percent complete. Bambu Handy also show live feed from camera

### Project file & Debug log uploads

Send file to print, jumps to camera, will not connect, shows network issue but updates layers and percent complete. Bambu Handy also show live feed from camera

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LavonT07, 2024-07-25:
Problem is intermittent.

Send file to print, jumps to camera, will not connect, shows network issue but updates layers and percent complete. Bambu Handy also show live feed from camera

bambu123, 2024-07-30:
@LavonT07 Please provider studio log.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log