bambulab/BambuStudio #4545: AMS not retracting

steven-l-heck, 2024-07-26:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S Combo

### How to reproduce

1. Through software or printer try to load fillament
2. Get an error that filament is not retracting

### Actual results

I attempt to load filament in all 4 slots of the AMS. The filament auto loads, but when I try to calibrate the printer or to print a job, I get an error that the spool is not retracting

### Expected results

Job should print

### Project file & Debug log uploads

[Printer config bundle.zip](https://github.com/user-attachments/files/16394328/Printer.config.bundle.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

steven-l-heck, 2024-07-26:
I tried checking the tubes, but cannot find any filament stuck in a line anywhere. I was able to print single color from it this morning, but when I tried adding the other three spools (all Bambu filament) it started giving an error.
[image]

steven-l-heck, 2024-07-26:
[image]

steven-l-heck, 2024-07-26:
I removed all filament, then reloaded the original color in the same slot (far left 1) and am now able to print.

steven-l-heck, 2024-07-26:
While that file was printing, the spool hung up and got stuck. I got the below error. I freed up the spool, pulled out the filament and reloaded, making sure the spool would spin freely. I got the same error.
[image]

steven-l-heck, 2024-07-26:
I loaded Orange, it works. I am able to use both spools that came with the printer. I also ordered half a dozen spools of various Basic PLA at the same time. These spools are larger and heavier. I have tried loading two of them. They feed in, but part of the load process is to back out and feed again. These larger rolls won't back out.

ttt888zzz, 2024-07-30:
HI，[steven-l-heck](https://github.com/steven-l-heck)：
To learn more about p1s, please visit：https://wiki.bambulab.com/en/hms/home
[image]