bambulab/BambuStudio #4546: Projects become locked to MakerWorld projects and cannot be unlinked

t0nyz0, 2024-07-26:
I printed something from MakerWorld and when it finished I cleared the STL out of BambuStudio. 
I then imported some other STLs in and started another project only to realize hours later that the Project tab is still tied to a the Makerworld print. **If the project tab is locked to a MakersWorld project then BambuStudio should not allow you to import other STLs**. This is a terrible bug, I wasted at least 2 hours just on this one instance. Since you cant clear what project its locked to you have to literally start all over again with a fresh BambuStudio instance. 
Bambu has likely cost me dozens of hours.

Also if you just double click on an STL on your desktop it will open into the active BambuStudio window, which could be a makersworld project. From that point on that 3MF/STL will forever be locked to that project. A very simple fix is just to have every STL open in a new window, especially if the current window is locked to a MakerWorld project.  You should not be able to double click and it open into the current project, it should start a new one.  If people want to open STLs to a current project they need to use the Import button. 

I think at this point Bambu either does not care about this issue or simply do not understand the problem and how to fix it. But either way, this is one bug that is so irritating that I'm tempted to never open another MakerWorld project again.

AT MINIMUM: When using "Export to Generic 3MF" it should strip all project information. Or give us a way to remove the the Project information all together, CLEAR button. Just basic edit ability. Not sure how this has lasted this long and this issue is marked as closed. Really disappointing. 

[image]
[image]

@bambu123 I dont think you guys are understanding the issue. We dont need to edit a lot of stuff. We need to remove the link between online MakerWord projects and the BambuStudio project that we are currently working on. Its often not correct and tied to old projects that are no longer pulled up.

_Originally posted by @t0nyz0 in https://github.com/bambulab/BambuStudio/issues/2111#issuecomment-2253473234_

t0nyz0, 2024-07-27:
After a few hours of testing I figured out that you can open a new Project and import in 3MF files that are associated with MakerWorld projects and it will remove the link. This is a hack to work around the problem. It still requires you to fix the plate layouts but it does mean I can get around it. I'll mark this as closed but if you guys would add a "UNLINK" or "CLEAR"  in the Project tab that would save me a lot of trouble.

closed 2024-07-27 by t0nyz0

KriyenKP, 2024-11-18:
Don't think you should have closed this. Even creating a new project doesn't let me unlink the embedded one.