bambulab/BambuStudio #4552: Bambu Studio crashes on launch on Ubuntu 24.04

0xjmux, 2024-07-29:
### Bambu Studio Version

1.9.03.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04 LTS

### Printer

NA

### How to reproduce

Launch Bambu studio with flatpak or Appimage on Ubuntu 24. 

### Actual results

The Application doesn't open, and produces the below log.  Prusaslicer launches without issue on my machine, so it's a bambu studio issue.
The process also hangs, and has to be killed manually; it gets stuck and refuses to respond to all process kill signals. 
``
user      626976  0.4  0.2 89529760 196328 ?     Sl   09:34   0:08 /tmp/.mount_Bambu_VfIHzF/bin/bambu-studio
user      627082  0.0  0.1 342944 81724 ?        Dl   09:34   0:00 /usr/lib/x86_64-linux-gnu/../../lib/x86_64-linux-gnu/gstreamer1.0/gstreamer-1.0/gst-plugin-scanner -l /tmp/.mount_Bambu_VfIHzF/bin/bambu-studio
user      627758  0.0  0.1 342948 81948 pts/14   D    09:34   0:00 /usr/lib/x86_64-linux-gnu/../../lib/x86_64-linux-gnu/gstreamer1.0/gstreamer-1.0/gst-plugin-scanner -l /tmp/.mount_Bambu_k5R7Ey/bin/bambu-studio
```

### Expected results

The application should open and not immediately crash

### Project file & Debug log uploads

```
~/Downloads  ./Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage
[2024-07-29 09:34:40.707142] [0x000075b7dcaecc80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/user/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/user/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:45.940: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:45.968: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:45.969: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:45.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:45.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:45.972: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:45.973: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI3: Screen seems not DRI3 capable

** (bambu-studio:627650): CRITICAL **: 09:34:46.048: Cannot register URI scheme wxfs more than once

** (bambu-studio:627650): CRITICAL **: 09:34:46.048: Cannot register URI scheme memory more than once

** (bambu-studio:627650): CRITICAL **: 09:34:46.050: Cannot register URI scheme wxfs more than once

** (bambu-studio:627650): CRITICAL **: 09:34:46.050: Cannot register URI scheme memory more than once

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.064: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.067: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.067: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.067: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.067: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.067: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.067: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.067: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.067: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.073: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.074: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.074: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.150: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.157: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.157: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:627650): Gtk-CRITICAL **: 09:34:46.157: gtk_widget_set_size_request: assertion 'width >= -1' failed
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-07-30:
```
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI3: Screen seems not DRI3 capable
```
Looks like a display driver problem. 
1. Are you using a nvidia GPU? try updating the drivers.
2. switch xorg window manager.

bschatz, 2024-08-24:
Maybe an issue with the network plugin?

~~~
loading network module at /home/user/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/user/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)
~~~

JumpMaster, 2024-09-14:
I have the same issue although I've caveat that I'm using ARM64.

```
lsloading network module at /home/jumpmaster/.config/BambuStudioInternal/plugins/libbambu_networking.so
error, dlerror is /home/jumpmaster/.config/BambuStudioInternal/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)
```

The file does exist.

Is the libbambu_networking.so file specifically built for X64?

MackBambu, 2024-09-20:
> I have the same issue although I've caveat that I'm using ARM64.
>
> ```
> lsloading network module at /home/jumpmaster/.config/BambuStudioInternal/plugins/libbambu_networking.so
> error, dlerror is [...]
The network plug-in does not yet support the arm platform.
https://github.com/bambulab/BambuStudio/issues/4658

MackBambu, 2024-09-20:
> I have the same issue although I've caveat that I'm using ARM64.

I have the same issue although I've caveat that I'm using ARM64.
https://github.com/bambulab/BambuStudio/issues/4658

jymbob, 2024-11-04:
Same issue here - unable to launch
* Ubuntu 24.04
* AMD x86_64
* AMD graphics
* X11 (not wayland)

Release:  v01.10.00.81 (latest at time of writing)

Logs:
```
$ ./Bambu_Studio_ubuntu-24.04_PR-5090.AppImage 
[2024-11-04 10:27:37.632004] [0x00007d2b72486400] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/development/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/development/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory after dlopen, network_module is (nil)

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.183: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.226: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.226: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.227: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.228: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.232: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.233: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (bambu-studio:28709): CRITICAL **: 10:27:47.327: Cannot register URI scheme wxfs more than once

** (bambu-studio:28709): CRITICAL **: 10:27:47.327: Cannot register URI scheme memory more than once

** (bambu-studio:28709): CRITICAL **: 10:27:47.330: Cannot register URI scheme wxfs more than once

** (bambu-studio:28709): CRITICAL **: 10:27:47.330: Cannot register URI scheme memory more than once

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.350: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.353: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.353: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.353: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.354: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.354: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.354: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.354: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.354: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.361: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.361: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:28709): Gtk-CRITICAL **: 10:27:47.361: gtk_widget_set_size_request: assertion 'width >= -1' failed
Killed
```

jymbob, 2024-11-04:
Hm, the path reported above: `/home/development/.config/BambuStudio/plugins/libbambu_networking.so` isn't there. In fact the entire `plugins` directory is missing

```
$ ll ~/.config/BambuStudio/
total 36
drwxrwxr-x  8 development development 4096 Nov  4 10:27 ./
drwx------ 95 development development 4096 Nov  4 10:06 ../
-rw-rw-r--  1 development development 2172 Nov  4 10:27 BambuStudio.conf
drwxrwxr-x  2 development development 4096 Nov  4 10:06 cache/
drwxrwxr-x  2 development development 4096 Nov  4 10:27 log/
drwxrwxr-x  2 development development 4096 Nov  4 10:07 ota/
drwxrwxr-x  2 development development 4096 Nov  4 10:07 printers/
drwxrwxr-x  3 development development 4096 Nov  4 10:27 system/
drwxrwxr-x  3 development development 4096 Nov  4 10:07 user/
```

Error in the Ubuntu build script?

zaakkniight-creator, 2025-09-20:
Hello everyone, SSL certificate error, juste click on **YES** and for language error, open the terminal and type the commande:
"**sudo locale-gen en_GB en_GB.UTF-8
sudo update-locale**"
It's worked for me.