bambulab/BambuStudio #4553: Bind with PIN code: Confirm button works not (Linux)

bschatz, 2024-07-29:
### Bambu Studio Version

1.9.3.50 (v01.09.03.50-20-gd59cb73a)

### Where is the application from?

Built from source by yourself or others

### OS version

Debian 12.5

### Additional system information

AMD Phenom(tm) II X6 1090T Processor
12 GB RAM
VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 23 [Radeon RX 6600/6600 XT/6600M] (rev c7)

### Printer

Bambu Lab P1S

### How to reproduce

1. device
2. "No printer" +
3. Bind with PIN Code
4. Enter the PIN Code displayed on the printer
5. Press "Confirm" Button 

### Actual results

Nothing happens after pressing the "Confirm"
windows with PIN Code is still open, no further messages.
Cancel Button closes the windows. 

### Expected results

The printer should be bind or at least a confirm message,
error message or any message should appear.

### Project file & Debug log uploads

[logs.tar.gz](https://github.com/user-attachments/files/16419062/logs.tar.gz)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Discobanan, 2024-07-31:
I have the same issue on MacOS using version 1.9.3.50 of Bambu Studio

Discobanan, 2024-07-31:
Nevermind, after logging in to my bambu-account first, it works and binds properly. Would be nice with some error-message saying "you need to login first" though, instead of nothing happening :)

bschatz, 2024-07-31:
yes, in the meantime i also saw that you have first to login.

ywiyogo, 2024-08-01:
I had the same issue. First login, then binding a device.
The bind and add device should be disabled if there is no user logged in.

walterwongbbl, 2024-08-05:
@bschatz 
Hi. We have reproduced this issue and will arrange time to fix it.

majodi, 2025-11-13:
Okay, so you have to login first. But why not show a message to the user?