bambulab/BambuStudio #4554: 切片崩溃

yourbask, 2024-07-30:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 专业版 64位（Version 23H2 / DirectX 12）

### Additional system information

  处理器                       英特尔 Core i9-14900K
  主板                         华硕 ROG MAXIMUS Z790 APEX ENCORE（LPC Controller/eSPI Controller Z790芯片组）
  显卡                         NVIDIA GeForce RTX 4090 ( 24 GB / 华硕 )
  内存                         48 GB ( 芝奇 DDR5 8000MHz 24GB x 2 )
  主硬盘                       三星 SSD 990 PRO 2TB (2 TB / 固态硬盘)
  显示器                       飞利浦 PHL092B PHL 329P9 ( 31.5 英寸  )
  声卡                         Generic USB Audio(Generic USB Audio)
  网卡                         英特尔 Wi-Fi 7 BE200 320MHz

### Printer

X1c

### How to reproduce

[image]

### Actual results

卡在这个界面然后 崩溃闪退

### Expected results

卡在这个界面然后 崩溃闪退

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16421654/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-07-31:
你好[yourbask](https://github.com/yourbask)，可以请您提供一下源文件吗

QingZhangBambu, 2024-08-01:
@yourbask 
https://www.xda-developers.com/intel-cracks-the-cause-raptor-lake-instability/
It's an intel software bug that's been going on for a while now.
Turning off the overclocking settings on the motherboard will improve it.

https://github.com/bambulab/BambuStudio/issues/3318
https://github.com/prusa3d/PrusaSlicer/issues/11845

yourbask, 2024-08-29:
> @yourbask https://www.xda-developers.com/intel-cracks-the-cause-raptor-lake-instability/ 这是英特尔软件错误，已经存在一段时间了。 关闭主板上的超频设置可以改善它。
> 
> [#3318 ](https://github.com/bambulab/BambuStudio/issues/3318) [prusa3d/PrusaSlicer#11845](https://github.com/prusa3d/PrusaSlicer/issues/11845)

已经确定是 13代 14代 i9处理器的问题，更新BIOS 0x129微码补丁,或主动降频降压均可解决此崩溃问题。

closed 2024-08-29 by yourbask