bambulab/BambuStudio #4555: Can't enter value in manual k-value calibration

Zydnar, 2024-07-30:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

1. Go to Calibration tab
2. Perform Flow Dynamics manual test
3. Try to enter k value either using `.` as at official Wiki or `,`
4. Click Finish button

### Actual results

[image]
[image]

### Expected results

Value in format from screenshot at official Wiki should work - see: https://wiki.bambulab.com/software/bambu-studio/calibration/pa_save_auto_results.jpg for reference.

### Project file & Debug log uploads

[test.zip](https://github.com/user-attachments/files/16428855/test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-08-01:
What is the language of your operating system?
And is the language of your Studio is set to English?

Zydnar, 2024-08-01:
@zhimin-zeng-bambulab You're suggesting number separator in settings there affects slicer? It could be - I changed default setting there to work with Fullcontrol XYZ macros https://fullcontrolgcode.com/download I'll confirm it and post an update...

closed 2024-08-01 by Zydnar