bambulab/BambuStudio #4556: Bambu Studio crashes after startup with "Illegal instruction"

bschatz, 2024-07-30:
### Bambu Studio Version

01.09.03.50

### Where is the application from?

Bambu Lab github releases

### OS version

Debian 12.5 (Bookworm)

### Additional system information

AMD Phenom(tm) II X6 1090T Processor
12GB
VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 23 [Radeon RX 6600/6600 XT/6600M] (rev c7)

### Printer

Bambu Lab P1S

### How to reproduce

* Start ./Bambu_Studio_linux_ubuntu-v01.09.03.50.AppImage 

### Actual results

Bambu Studio is a short time visible then it crashes
Output on console:

```
BambuTunnel::GetMsg(1)
Illegal instruction
```

### Expected results

app crashes not

### Project file & Debug log uploads

[logs.tar.gz](https://github.com/user-attachments/files/16429883/logs.tar.gz)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bschatz, 2024-07-31:
sorry, i forgot the backtrace:

```
(gdb) backtrace
#0  0x00007fffcf783077 in TransCodePartial () at /home/bernd/.config/BambuStudio/plugins/libBambuSource.so
#1  0x00007fffcf78870d in iotc_SendMessage () at /home/bernd/.config/BambuStudio/plugins/libBambuSource.so
#2  0x00007fffcf78bdaf in _IOTC_SendQuryDevice5 () at /home/bernd/.config/BambuStudio/plugins/libBambuSource.so
#3  0x00007fffcf79d4a4 in AddUDPQueryDeviceInfoTask () at /home/bernd/.config/BambuStudio/plugins/libBambuSource.so
#4  0x00007fffcf7a355c in IOTC_Connect_UDP_Inner () at /home/bernd/.config/BambuStudio/plugins/libBambuSource.so
#5  0x00007fffcf7a3ce8 in IOTC_Connect_ByUIDEx () at /home/bernd/.config/BambuStudio/plugins/libBambuSource.so
#6  0x00007fffcf74f3a8 in BambuTunnelTutk::open() () at /home/bernd/.config/BambuStudio/plugins/libBambuSource.so
#7  0x0000555557158081 in gst_bambusrc_start ()
#8  0x00007ffff03c4c3a in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#9  0x00007ffff03c51d8 in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#10 0x00007ffff6ab8fd9 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#11 0x00007ffff6ab9750 in gst_pad_set_active () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#12 0x00007ffff6a92121 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#13 0x00007ffff6aa8c74 in gst_iterator_fold () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#14 0x00007ffff6a927a6 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#15 0x00007ffff6a94a5a in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#16 0x00007ffff6a94e18 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#17 0x00007ffff03c2c8a in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#18 0x00007ffff6a96f30 in gst_element_change_state () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#19 0x00007ffff6a96f7b in gst_element_change_state () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#20 0x00007ffff6a9759b in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#21 0x00007fffcc50dd01 in  () at /lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so
#22 0x00007ffff6a96f30 in gst_element_change_state () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#23 0x00007ffff6a96f7b in gst_element_change_state () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#24 0x00007ffff6a9759b in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#25 0x00007fffcc52c78e in  () at /lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so
#26 0x00007fffcc52d5aa in  () at /lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so
#27 0x00007ffff6a96f30 in gst_element_change_state () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#28 0x00007ffff6a96f7b in gst_element_change_state () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#29 0x00007ffff6a9759b in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#30 0x00005555591fa849 in wxGStreamerMediaBackend::DoLoad(wxString const&) ()
#31 0x00005555591fac7b in wxGStreamerMediaBackend::Load(wxURI const&) ()
#32 0x00005555591f4ba2 in wxMediaCtrl::Load(wxURI const&) ()
#33 0x0000555557156c40 in wxMediaCtrl2::Load(wxURI) ()
#34 0x00005555570c565b in Slic3r::GUI::MediaPlayCtrl::media_proc() ()
#35 0x000055555755675b in thread_proxy ()
#36 0x00007ffff0223134 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#37 0x00007ffff02a37dc in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

bschatz, 2024-08-01:
it is installed:

```
$ sudo aptitude search gstreamer1.0-x
i A gstreamer1.0-x                                - GStreamer plugins for X11 and Pango                    
i A gstreamer1.0-x:i386                           - GStreamer plugins for X11 and Pango 
```

berndschatz, 2024-08-02:
in the meantime i have a also upgraded to debian 12.6, but same issue.
I will try it from an other debian 12.6 system or with help of a live system again.
strange ...

MackBambu, 2024-08-03:
Or you can try using the flatpak version instead.
https://flathub.org/apps/com.bambulab.BambuStudio

bschatz, 2024-08-20:
I have tested the AppImage now on three different Hardware (Intel/AMD CPU and Nvidia/AMD GPU) on Debian 12.6, same result.
It crashes also on a **Ubuntu 24.04 LTS** in a virtual machine (qemu/KVM).

1. Start AppImage
2. Question about Cert --> Yes (keep the decision)
3. Restarted
4. Setup Wizard
4.1 Get Started
4.2 Europe
4.3 "Join the program" -> Skip
4.4 Printer Selection --> Clear All, choose Bambu Lab P1S 0.4mm nozzle
4.5 Filament Section --> Clear all, choose Bambu PLA Basic and Bambu Support for PLA
4.6 Install Bambu Network plugin-in
5. Login with choosing Google Account (accept all cookies)  
5.1 Synchronization ? --> Yes
6 Klick on Menu "Device" -->  

```
BambuTunnel::GetMsg(1)
Illegal instruction
```

bschatz, 2024-08-28:
> My tests on Debian 12.6 work fine. From your message it seems to be a gstreamer issue, and since similar problems have occurred on other debain distributions, I suggest you install the gstreamer 1.0-x package to try. apt install gstreamer1.0-x

The idea of AppImage is that they run on all distribution by including all necessary dependencies ,
but if the plugins depends on libs from the system it makes no sense to use AppImage.
Why not just offer dep packages where the dependencies are well defined ?

wabuMike, 2024-09-05:
Same error here with the same message. Also on Ubuntu 20.04.

wabuMike, 2024-09-05:
The app crashes when the printer is connected to the network. If no printer is connected, then the app runs.

wabuMike, 2024-09-07:
@MackBambu I've now tried flatpak, Docker and self-compiled version with a very similar setup as described above (same hardware and OS). They all crash with the message described above as soon as the BambuLab printer connects to the network. No printer, no problem - the Studio app runs.

bschatz, 2024-09-14:
> @MackBambu I've now tried flatpak, Docker and self-compiled version with a very similar setup as described above (same hardware and OS). They all crash with the message described above as soon as the BambuLab printer connects to the network. No printer, no problem - the Studio app runs.

Yes, i can confirm that.
If i understand it correctly, it is the network plugin.

* Is it possible to increase the log level of the plugin ?
* The source code of the this plugin is not in the bambu studio repo ?

MackBambu, 2024-09-14:
> Yes, i can confirm that. If i understand it correctly, it is the network plugin.
> 
>     * Is it possible to increase the log level of the plugin ?
> 
>     * The source code of the this plugin is not in the bambu studio repo ?

The network plugin code is non-open source, so I'll try to see if I can reproduce the problem.

saidlm, 2024-11-20:
Hi,
Is there any progress? I am experiencing pretty same issue on Kubuntu 24.04 with version 1.10.1. As soon as the printer is on (on-line) the application crashes almost immediately.

```
BambuTunnel::GetMsg(1)
Illegal instruction (core dumped)
```

Thank you for any help with that, Martin.

bschatz, 2025-05-15:
The network plugin uses obviously AVX code:

~~~
sudo apt install llvm
cd ~/.config/BambuStudio/plugins
llvm-objdump -d libbambu_networking.so | grep -E '\bv[a-z]{2,}' | grep vpxor
  28029f: c5 15 ef 07                  	vpxor	(%rdi), %ymm13, %ymm8
  357a5a: c5 51 ef 64 17 78            	vpxor	120(%rdi,%rdx), %xmm5, %xmm12
  38234c: c5 fd ef d0                  	vpxor	%ymm0, %ymm0, %ymm2
~~~

All my devices i see the crash , the cpu does not support avx.
On devices with cpu that supports avx it works fine.
You can verify wether your cpu supports avx with: 

~~~
lscpu | grep avx
~~~

My suggestion:
(1) If you use avx you should verify wether the cpus supports that 
(2) Better: Offer the networkplugin with avx disabled.

bschatz, 2025-06-20:
Hi @saidlm, @MackBambu 

can you verify my assumption, that the error belongs only
to CPUs that do not support the AVX extension ?

closed 2025-08-15 by bschatz