bambulab/BambuStudio #4557: keeps crashing

raidersc, 2024-07-30:
### Bambu Studio Version

1.9.35

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

System Information
------------------
      Time of this report: 7/30/2024, 14:57:24
             Machine name: DESKTOP-KS99E4T
               Machine Id: {D30A4DEE-BD4B-483E-A8A4-5256DD28B696}
         Operating System: Windows 11 Home 64-bit (10.0, Build 22631) (22621.ni_release.220506-1250)
                 Language: English (Regional Setting: English)
      System Manufacturer: Micro-Star International Co., Ltd.
             System Model: MS-7D75
                     BIOS: 1.70 (type: UEFI)
                Processor: AMD Ryzen 7 7800X3D 8-Core Processor            (16 CPUs), ~4.2GHz
                   Memory: 65536MB RAM
      Available OS Memory: 64672MB RAM
                Page File: 38453MB used, 73323MB available
              Windows Dir: C:\Windows
          DirectX Version: DirectX 12
      DX Setup Parameters: Not found
         User DPI Setting: 168 DPI (175 percent)
       System DPI Setting: 144 DPI (150 percent)
          DWM DPI Scaling: UnKnown
                 Miracast: Available, no HDCP
Microsoft Graphics Hybrid: Supported
 DirectX Database Version: 1.5.8
           DxDiag Version: 10.00.22621.3527 64bit Unicode

------------
DxDiag Notes
------------
      Display Tab 1: No problems found.
      Display Tab 2: No problems found.
      Display Tab 3: No problems found.
      Display Tab 4: No problems found.
        Sound Tab 1: No problems found.
        Sound Tab 2: No problems found.
        Sound Tab 3: No problems found.
        Sound Tab 4: No problems found.
        Sound Tab 5: No problems found.
        Sound Tab 6: No problems found.
        Sound Tab 7: No problems found.
        Sound Tab 8: No problems found.
        Sound Tab 9: No problems found.
          Input Tab: No problems found.

--------------------
DirectX Debug Levels
--------------------
Direct3D:    0/4 (retail)
DirectDraw:  0/4 (retail)
DirectInput: 0/5 (retail)
DirectMusic: 0/5 (retail)
DirectPlay:  0/9 (retail)
DirectSound: 0/5 (retail)
DirectShow:  0/6 (retail)

---------------
Display Devices
---------------
           Card name: NVIDIA GeForce RTX 4060 Ti
        Manufacturer: NVIDIA
           Chip type: NVIDIA GeForce RTX 4060 Ti
            DAC type: Integrated RAMDAC
         Device Type: Full Device (POST)
          Device Key: Enum\PCI\VEN_10DE&DEV_2803&SUBSYS_13EC196E&REV_A1
       Device Status: 0180200A [DN_DRIVER_LOADED|DN_STARTED|DN_DISABLEABLE|DN_NT_ENUMERATOR|DN_NT_DRIVER] 
 Device Problem Code: No Problem
 Driver Problem Code: Unknown
      Display Memory: 48445 MB
    Dedicated Memory: 16109 MB
       Shared Memory: 32336 MB
        Current Mode: 3840 x 2160 (32 bit) (60Hz)
         HDR Support: Not Supported
    Display Topology: Extend
 Display Color Space: DXGI_COLOR_SPACE_RGB_FULL_G22_NONE_P709
     Color Primaries: Red(0.633789,0.340820), Green(0.311523,0.635742), Blue(0.158203,0.061523), White Point(0.312500,0.329102)
   Display Luminance: Min Luminance = 0.500000, Max Luminance = 270.000000, MaxFullFrameLuminance = 270.000000
        Monitor Name: Generic PnP Monitor
       Monitor Model: U28E510
          Monitor Id: SAM0D63
         Native Mode: 3840 x 2160(p) (59.997Hz)
         Output Type: Displayport External

### Printer

X1c 

### How to reproduce

try to load and slice a file, then when loading to printer sometimes just randomly crashes 

### Actual results

program closes 

### Expected results

should start printing. 

### Project file & Debug log uploads

[
[crash_Tue_Jul_30_14_49_05_0_done.log](https://github.com/user-attachments/files/16432899/crash_Tue_Jul_30_14_49_05_0_done.log)
[crash_Tue_Jul_30_14_49_41_0_done.log](https://github.com/user-attachments/files/16432900/crash_Tue_Jul_30_14_49_41_0_done.log)
[crash_Tue_Jul_30_14_50_23_0_done.log](https://github.com/user-attachments/files/16432901/crash_Tue_Jul_30_14_50_23_0_done.log)
[crash_Tue_Jul_30_14_51_05_0_done.log](https://github.com/user-attachments/files/16432902/crash_Tue_Jul_30_14_51_05_0_done.log)
[crash_Tue_Jul_30_14_51_37_0_done.log](https://github.com/user-attachments/files/16432903/crash_Tue_Jul_30_14_51_37_0_done.log)
[crash_Tue_Jul_30_14_54_34_0_done.log](https://github.com/user-attachments/files/16432904/crash_Tue_Jul_30_14_54_34_0_done.log)
[crash_Sun_Jul_28_15_42_22_0_done.log](https://github.com/user-attachments/files/16432905/crash_Sun_Jul_28_15_42_22_0_done.log)
](url)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ttt888zzz, 2024-07-31:
HI，[raidersc](https://github.com/raidersc)
Could you please provide a screenshot or video of the operation procedure，We'll check it.

raidersc, 2024-07-31:
I actually uninstalled it last night then deleted all folders. Downloaded software again then installed it. Then had an issue with a plugin installing so deleted plugin folder. Then relaunched program and plugin installed. Now it’s working fine again so far no crashes. Sent from my iPhone

raidersc, 2024-08-01:
It’s back to just closing the app again  I restart my pcAnd the printer the. The file will print. Sent from my iPhone

walterwongbbl, 2024-08-01:
@raidersc 
Is everything alright now?

raidersc, 2024-08-01:
It seems to be hit and miss. It’s not the stl. I tried switching to a different file and it would still crash and close. I’m still trying to isolate what causes it and what fixes it so far it seems like a computer restart and a printer restart will fix the problem.Sent from my iPhone

ttt888zzz, 2024-08-02:
Can you please try the following steps, it may help：
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

raidersc, 2024-08-02:
I’ll try it but it doesn’t crash during slicing it crashes while sending to printer. Sent from my iPhone

https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

raidersc, 2024-08-03:
Set affinity. Still closes when trying to print. I’ve isolated the problem to be a communication with printer issue. I restart printer and it will print again. Sent from my iPhone

https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing