bambulab/BambuStudio #4559: Segmentation fault on MX Linux 23 using the latest AppImage

ywiyogo, 2024-07-31:
### Bambu Studio Version

1.9.3

### Where is the application from?

Bambu Lab github releases

### OS version

MX Linux 23

### Printer

Bambu Lab P1S

### How to reproduce

1. Download the App Image from the Github release
2. set the execute permission
3. Run the AppImae on the terminal
4. 

### Actual results

Segmentation fault.
After I clone and build the repo with the debug version, run with gdb, this is the backtrace result:
```
Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x0000555557b9d6d9 in wxMediaCtrl2::wxMediaCtrl2 (this=0x55555ef949c0, parent=0x55555efe1300)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/wxMediaCtrl2.cpp:42
42	    auto playbin = reinterpret_cast<wxGStreamerMediaBackend *>(m_imp)->m_playbin;
(gdb) 
(gdb) bt
#0  0x0000555557b9d6d9 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*)
    (this=0x55555ef949c0, parent=0x55555efe1300)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/wxMediaCtrl2.cpp:42
#1  0x0000555557afac3b in Slic3r::GUI::StatusBasePanel::create_monitoring_page()
    (this=0x55555efe1300) at /home/li/apps/BambuStudio/src/slic3r/GUI/StatusPanel.cpp:990
#2  0x0000555557af6d60 in Slic3r::GUI::StatusBasePanel::StatusBasePanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&)
    (this=0x55555efe1300, parent=0x55555edf4170, id=-1, pos=..., size=..., style=524288, name=...)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/StatusPanel.cpp:825
#3  0x0000555557b039ae in Slic3r::GUI::StatusPanel::StatusPanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&)
    (this=0x55555efe1300, parent=0x55555edf4170, id=-1, pos=..., size=..., style=524288, name=...)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/StatusPanel.cpp:1623
#4  0x00005555573dce16 in Slic3r::GUI::MonitorPanel::init_tabpanel() (this=0x55555ef74c00)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/Monitor.cpp:211
#5  0x00005555573db283 in Slic3r::GUI::MonitorPanel::MonitorPanel(wxWindow*, int, wxPoint const&, wxSize const&, long) (this=0x55555ef74c00, parent=0x55555e1b0800, id=-1, pos=..., size=..., style=524288)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/Monitor.cpp:107
```

### Expected results

No segmentation fault

### Project file & Debug log uploads

Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x0000555557b9d6d9 in wxMediaCtrl2::wxMediaCtrl2 (this=0x55555ef949c0, parent=0x55555efe1300)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/wxMediaCtrl2.cpp:42
42	    auto playbin = reinterpret_cast<wxGStreamerMediaBackend *>(m_imp)->m_playbin;
(gdb) 
(gdb) bt
#0  0x0000555557b9d6d9 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*)
    (this=0x55555ef949c0, parent=0x55555efe1300)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/wxMediaCtrl2.cpp:42
#1  0x0000555557afac3b in Slic3r::GUI::StatusBasePanel::create_monitoring_page()
    (this=0x55555efe1300) at /home/li/apps/BambuStudio/src/slic3r/GUI/StatusPanel.cpp:990
#2  0x0000555557af6d60 in Slic3r::GUI::StatusBasePanel::StatusBasePanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&)
    (this=0x55555efe1300, parent=0x55555edf4170, id=-1, pos=..., size=..., style=524288, name=...)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/StatusPanel.cpp:825
#3  0x0000555557b039ae in Slic3r::GUI::StatusPanel::StatusPanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&)
    (this=0x55555efe1300, parent=0x55555edf4170, id=-1, pos=..., size=..., style=524288, name=...)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/StatusPanel.cpp:1623
#4  0x00005555573dce16 in Slic3r::GUI::MonitorPanel::init_tabpanel() (this=0x55555ef74c00)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/Monitor.cpp:211
#5  0x00005555573db283 in Slic3r::GUI::MonitorPanel::MonitorPanel(wxWindow*, int, wxPoint const&, wxSize const&, long) (this=0x55555ef74c00, parent=0x55555e1b0800, id=-1, pos=..., size=..., style=524288)
    at /home/li/apps/BambuStudio/src/slic3r/GUI/Monitor.cpp:107

### Checklist of files to include

- [X] Log file
- [ ] Project file

ywiyogo, 2024-08-01:
Note, in MX Linux, there is no libgstreamer-plugins-good1.0-dev, instead I installed libgstreamer-plugins-bad1.0-dev.

MackBambu, 2024-08-01:
hi, @ywiyogo Try installing this package.
apt install gstreamer1.0-x