bambulab/BambuStudio #4560: More than 10 objects splitted from a step files

yreichli, 2024-07-31:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows  365

### Printer

X1 carbon

### How to reproduce

1. add a step files with written elements on it ( for example : abcd-012345)
2. split the step files into objects or parts 
3. select different colors between the main part and the text

### Actual results

all the text is not included into the parts/Objects. 
Seems that this is limited to 10 parts? 
The objects and parts are automatically numbered, on a random way (from 1 to 9) (= if you do the test 3 times, the number x will be systematically a different part)

### Expected results

all objects should be visible

### Project file & Debug log uploads

[Bug Report.zip](https://github.com/user-attachments/files/16439218/Bug.Report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-07-31:
Hello, can you record a reproduction video or GIF？thanks

yreichli, 2024-07-31:
[image]

yreichli, 2024-07-31:
I split my step into objects or parts (here, parts), and the studio skip a caracter (here : should be Drop 100 instead of Drop 1 0)

Sometimes it is another caracter (not always the 0)

Haidiye00, 2024-08-01:
Hello, can you provide this file? Thank you? Facilitating our analysis of the reasons

MackBambu, 2024-10-18:
https://github.com/bambulab/BambuStudio/commit/90f325792dc4910b80d69c9d6a5677f24b8fa259
This commit should fix the issue, if you still have questions you can reopen it.

closed 2024-10-18 by MackBambu