bambulab/BambuStudio #4565: Device Update tab is Blank

julie777, 2024-07-31:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1P or A1mini

### How to reproduce

1. open bambu studio
2. go to device tab
3. select printer
4. select update on the left

Note: I run my printers in LAN only mode

### Actual results

The entire right pane is blank

### Expected results

Should show the firmware version for the printer like it used to.

In addition, I would like it to allow updating printer firmware from bambu studio so that I do not have to connect my printer to the internet.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-08-01:
HI，[julie777](https://github.com/julie777)
Could you please provide a screenshot，we will check it。

julie777, 2024-08-02:
[image]

eolder, 2024-08-14:
Can confirm this behavior with lan mode on P1S.

eolder, 2024-08-14:
> In addition, I would like it to allow updating printer firmware from bambu studio so that I do not have to connect my printer to the internet.

This is also important to me, please consider this feature!

MickGyver, 2024-11-02:
Same problem still on version 1.9.7.52

OnSive, 2025-04-16:
Any news on this one?

mrakus-sofomo, 2025-05-03:
2.0.3.54 on MacOS, issue still occurs