bambulab/BambuStudio #4566: Bambu Studio prevents Windows Screen Lock

julie777, 2024-07-31:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

p1p

### How to reproduce

1. open bambu studio
2. select device tab
3. select printer

### Actual results

Windows 10 does not lock the screen when the time limit expires

### Expected results

Windows should lock the screen when the screen lock timeout expires

Note: I used to think it had to do with the video player running, but this time I was not viewing printer video.
After posting my last bug about the device update page being blank, I left Bambu Studio on the device update page and went out and worked in the yard. When I came back in an hour later my screen was not locked. My lock screen is set at 30 minutes.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-08-01:
It may also be that other applications are blocking the screen from standby.

bambu123, 2024-08-01:
Or there are printer errors, warning pop-ups

julie777, 2024-08-02:
@bambu123 
There are no open dialogs and there are no other applications preventing it. If I set my lock time to  2 minutes and then wait it locks. Then I start bambu studio as described and wait for 2 minutes and it does not lock.

My theory is that it has something to do with the way that Bambu Studio is using Microsoft Edge Webview.