bambulab/BambuStudio #4568: Align to Y-Axis no longer works in  1.9.3.50

K2Kevin, 2024-07-31:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Additional system information

M2 Mac

### Printer

Bambu X1C

### How to reproduce

Select Arrange, Align to Y axis, no results.

### Actual results

Nothing happens

### Expected results

Align objects in the center from front to back along Y axis evenly

### Project file & Debug log uploads

NA 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-08-01:
Can you record a video or GIF? Thank you. And provide the arranged files

tangketan, 2024-08-02:
I've got a 3mf of similar behavior and fixed the issue. Please wait for the next release.

closed 2025-04-16 by Haidiye00