bambulab/BambuStudio #4569: The filament run out now filament is rolled under the clear cove where I inserted new filament

czecht, 2024-08-01:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu X1C

### How to reproduce

1) black PLA run out  2) Inserted the new spool of black PLA (same brand and same everything)  3) filament inside the clear area where you feed the filament. 4) red flashing light #2 position where the black PLA was.
[image]
[image]

### Actual results

I can't use any other slots, like when I put the black PLA in the first slot!
I can hear a motor spinning for several minutes. It is the second slot motor, I think (I think it is called an extruder?). 

### Expected results

It  should feed the new spool PLA black and continue printing, like it did before.
I need help ASAP!  Tony Fleming, 402-477-3569  or my cell# 402-304-9650

### Project file & Debug log uploads

Here are the zip files: 
[TonysPooperCatcher.zip.7z.zip](https://github.com/user-attachments/files/16458720/TonysPooperCatcher.zip.7z.zip)
[debug_Sun_Jul_28_09_20_56_16572.log.1.zip](https://github.com/user-attachments/files/16458973/debug_Sun_Jul_28_09_20_56_16572.log.1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gaolegao-lx, 2024-08-02:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

czecht, 2024-08-02:
I fixed it by myself.
Stuff you needed wasn't there and or a previous few days ago, rather than
8/1/2024!
Second, the size was over 32GB so I included what would fit into your email!
I think that you want so much information at once, it isn't perty for a
customer like me! I'm 69, not 13!
You need a much simpler version for elderly people like me!
I would like to have a phone number and an email address to communicate
with technical support please.

The printer is AMAZING, the support is USELESS for me!
You MUST make the requirements easier and less technical!
Also, if the printer is wireless and it is connected to the internet, you
can peek into the printer and get all the info you need!
Or you can send a request to a customer and when they click it, the local
network will send you all that info - how about that?

Please make technical support easier for us, older people. Also, there are
many people who are not equipped with technical know-how!
I'm sure you all are very skilled and younger or young people, who don't
understand us, older people. So you should support the people with a
disposal income to afford your product,

Thank you very much.

Tony Fleming
1939 N 67th Street
Lincoln, NE 68505

***@***.***