bambulab/BambuStudio #4572: PETG-HF Not Recognized

wysguydev, 2024-08-02:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Edition	Windows 11 Pro for Workstations Version	22H2 Installed on	‎10/‎5/‎2022 OS build	22621.3880 Experience	Windows Feature Experience Pack 1000.22700.1020.0

### Additional system information

Device name	DESKTOP-SNKD94D
Processor	Intel(R) Xeon(R) E-2186M  CPU @ 2.90GHz   2.90 GHz
Installed RAM	64.0 GB (63.8 GB usable)
Device ID	C0138E25-DA5A-4667-A50C-B14A0550EB48
Product ID	00391-70000-00000-AA308
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
Nvidia Quadro P3200

### Printer

Bambu Lab Carbon X1

### How to reproduce

After latest update on Bambu Studio . . . PETG-HF is no longer recognized.  It is recognized as PETG instead.

### Actual results

Prints work but not sure system is using best parameters for PETG-HF

### Expected results

Expect PETG-HF to be recognized automatically by system.

### Project file & Debug log uploads

No File

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-08-05:
@wysguydev  Do you mean that the PA profile of the ams slot cannot be recognized normally for PETG-HF?
Could you describe your problem in more detail?

wysguydev, 2024-08-05:
Hello,

 

Prior to doing a configuration update on Bambu Studio, first slot was recognized as PETG-HF.  After configuration update . . . first slot is only recognized as PETG.

 

All the best,

 

Max P. Henzi

CEO

WysGuy Designs, LLC

307.248.0118

zeminator, 2024-08-23:
Not totally sure all, because I have been playing only few days (first 3D printer)... And from the start I was checking stuff from the Bambu Handy... 

After the latest firmware update for P1S combo, Bambu PETG-HF filament was not recognized (slot A1), in Windows Bambu Studio 1.9.3.50. 

I Did notice that PC version System Filaments the Bambu PETG-HF is totally missing from the list. Handy I can select PETG-HF but after that PC version doesn't know it at all...
Syncing from AMS doesnt bring any filament in A1
[image]

Device shows the filament but not correct color etc (filament is black).
[image]