bambulab/BambuStudio #4573: Problem Manual Flow Calibration in Bambu Software

Raffa-91, 2024-08-02:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

P1S

### How to reproduce

1. Perform a manual flow calibration of the filament.
2. Enter an incorrect value, e.g., 0.2 instead of 0.02.
3. Notice that the value is handled and set to 0.200 without any warning or error.
4. Change the value manually with "manage the results" to 0.02
5. Same Error 

### Actual results

Subsequent prints were of poor quality.The AMS filament view showed discrepancies in the values, specifically missing a 0 after the decimal point.

### Expected results

1. Value handle should reflect/check the exact value entered.
2. The Software should prompt an error or warning when an unusually high value is input.

### Project file & Debug log uploads

NotNeeded

### Checklist of files to include

- [X] Log file
- [ ] Project file

Raffa-91, 2024-09-06:
@LiZ-Li-BBL, @ttt888zzz

JUST FOR THE IMPORTANCE:

I encountered an issue during manual filament flow calibration where I mistakenly entered 0.2 instead of 0.02. The system accepted the value and set it to 0.200, leading to poor print quality.

I noticed that the values in the AMS filament view differed significantly, with a missing zero after the decimal point.

To correct this, I opened the calibration tab, managed the results, and set the value to 0.02. However, the issue persisted. I double-checked the AMS values, which correctly showed 0.02. After restarting and resetting everything without success, I reviewed the values again:

- For PLA Silver, the value was 0.0150
- For PLA Black, the value was 0.02

The solution was ultimately to set the value to 0.020.

PLEASE CHECK THE HANDLE OF THIS VALUE

ttt888zzz, 2024-09-06:
HI [Raffa-91](https://github.com/Raffa-91) ：
Please try to update the latest version，[](https://www.bambulab.cn/zh-cn/download/studio)。
[image]

Could you please provide the source file，we will check it.