bambulab/BambuStudio #4574: Items "joined" in Blender cannot received supports...

OriginalSyn5150, 2024-08-02:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Sonoma 14.5

### Additional system information

M3 Ultra chip

### Printer

Bambu A1

### How to reproduce

When in Blender using the "join" or "command + J" to form a single entity for multiple small parts then exporting to an STL, for some reason 
[image]

### Actual results

Some items may require supports, when you use the join in Blender the slicer does not render everything.  You can see the difference in these screenshots

[image]
[image]

### Expected results

I am not an expert in 3D software, but my expectation would be that the slicer does not see them as "different". He is a screenshot in Blender (4.1)
[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16474584/log.zip)
[STC-127.3mf.zip](https://github.com/user-attachments/files/16474585/STC-127.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-08-06:
The normal vector of the facets in your STL model is not facing outward, which can be fixed by right clicking. Or you can check how Blender software exports the correct STL file.
[image]

Haidiye00, 2024-10-09:
hello,you can use https://www.formware.co/onlinestlrepair this web to repair model.

closed 2024-10-16 by Haidiye00