bambulab/BambuStudio #4581: Arrange items currently on build plate, or by selected items, not all items in the project

SNDR-Tech, 2024-08-04:
I've noticed that the arrange function will arrange all items in the project. If I already have several build plates in the project I do not want to arrange all items in the project automatically. This is especially an issue when using multiple built plates in a project. I use multiple build plates to print parts by color. Which is more efficient and faster than having it color change with the AMS. Current I do it by manually dragging parts to different build plates. But it doesn't line up as clean or as efficient as can be or as packed as it can be per build plate.

I would like Bambu Studio to understand that parts already on other build plates in the project shouldn't be selected to be added to the active working build plate when using the arrange function. Or be able to select items to be arranged on the active build plate. Another 

I've merged the parts by color and then arranged the items. But if the parts are small they will all arrange on the same build plate. I've also made different projects and added the items to be printed in the different colors, then arranged and merged them and saving the 3fm. After that I've dragged and dropped those 3fm files on a new project with the other parts that are arranged and merged as well on a different build plate. It would be nicer to just work in a single project instead of making "sub assemblies" and them adding them all to a single project.

AbeFM, 2024-08-06:
I've thought ctrl-a should select everything on a given plate.

SNDR-Tech, 2024-08-06:
ctrl-a selects all. All items in the project even the ones on different plates

> I've thought ctrl-a should select everything on a given plate.