bambulab/BambuStudio #4582: Tree support bug printing mid air (Cone shapes)

DamianBan123, 2024-08-04:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD Ryzen 5600x, AMD 6900 XT, 32 GB 3600

### Printer

Bambu Lab A1

### How to reproduce

Get Nobody - The Thinker model, activate tree support, change the "tree support branch diameter" from default 2 mm to 6 mm (or bigger). Slice and look under models belly. You will see that middle of supports start printing mid-air

### Actual results

[image]
[image]
[image]

### Expected results

There should be grid pattern under the first layer of support roof, bridging between trunk walls, to support roof.

### Project file & Debug log uploads

[BishopBUG.zip](https://github.com/user-attachments/files/16487061/BishopBUG.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-08-08:
Yes, it's a known issue. The tree branch diameter can't be too big in this case. We'll continue to improve on this, but the solution won't be too soon.

Gofven, 2026-02-20:
> Yes, it's a known issue. The tree branch diameter can't be too big in this case. We'll continue to improve on this, but the solution won't be too soon. For now, you can change the support base pattern to anything but hollow. The infill structure will prevent printing support in the air.

I want to point out that setting the support pattern seems to have no effect on many cases where support will generate in mid air