bambulab/BambuStudio #4586: 部分被导入的svg文件产生非流形边，即使使用修复功能也无法恢复正确外观

moe001, 2024-08-05:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Additional system information

无关

### Printer

p1sc

### How to reproduce

1. 新建项目
2. 文件-导入 选择附件的svg文件
3. (可选)使用右下角的修复功能修复模型

### Actual results

svg文件中包含15个相同的汉字，可以观察到有一些字产生了破损。即使是使用了修复模型功能，它们的构造仍然不正确。
[image]

### Expected results

产生正确的导入结果（即15个字均是相同且完整的）

参照：来自openscad的挤出结果，15个字完全正确
[image]

### Project file & Debug log uploads

[result.zip](https://github.com/user-attachments/files/16490821/result.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-08-05:
i will fix it~

Haidiye00, 2024-10-16:
The new version has been released, please download and use it~

closed 2024-10-16 by Haidiye00