bambulab/BambuStudio #4588: Wrong slicing result for SVG of qr coede

leoleoasd, 2024-08-05:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu P1S

### How to reproduce

see my file

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[qr code.zip](https://github.com/user-attachments/files/16500821/qr.code.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

leoleoasd, 2024-08-05:
SVG file is edited and exported from figma; 3mf file is created using makeMySign on makerworld. 
Maybe a dup of #4586 and #2883.

QingZhangBambu, 2024-08-13:
@leoleoasd fix the mesh here, then the result would be right
[image]
[image]

Haidiye00, 2024-08-13:
The function is being optimized, please wait

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~