bambulab/BambuStudio #4593: Select available and compatible printers by default

AbeFM, 2024-08-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I run a number of very modest print farms (2-5 machines), and regularly run them in parallel.
[image]
After slicing, when I Print a file, the software is likely to suggest an incompatible printer and then complain to me that it isn't compatible.
This may help you to catch errors, but the vast majority of the time it just slows down what it an obvious decision, making the process more tedious.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
If there is an available printer in the pool which IS compatible and has the right material, etc, then I suggest that printer be automatically selected. A *passive* onscreen indicator could notify the user of the change.

[image]

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

SaltWei, 2024-08-08:
Good suggestion!

SaltWei, 2024-08-08:
用户有多个机型的时候，发送打印的时候，默认应该自动为用户选择匹配的机器。匹配的机器有多个的时候，应该默认选择idle的

AbeFM, 2024-08-08:
> 用户有多个机型的时候，发送打印的时候，默认应该自动为用户选择匹配的机器。匹配的机器有多个的时候，应该默认选择idle的 @walterwongbbl

You could even go a step further, after that, and filter on available materials/nozzle/etc.
Looking forward to it. Thanks!

AbeFM, 2026-09-18:
Is there any progress on this one? It is a constant annoyance.