bambulab/BambuStudio #4595: Unable to Send or Send All unless in LAN-only mode

Erin-Mounts, 2024-08-06:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.5

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Make sure printer is turned on and is not in LAN-only mode
2. Make sure there's plenty of room on the SD card
3. Make sure Bambu Studio is connected to the printer
4. Open a new project
5. Add and arrange .stl files
6. Slice
7. Select Send or Send All
8. Click Send

### Actual results

I get a pop up dialog titled "LAN Connection Failed (Sending  print file)" with instructions for how to troubleshoot printer connection issues. (I know it's not actually a printer connection issue because selecting "Print" instead of "Send" or "Send All" works just fine. Also for what it's worth, if I switch to LAN-only mode, I am then able to Send or Send All to the printer so I know it's not an SD card out of space issue.)

### Expected results

File should be uploaded to the SD card on the printer and be ready for printing.

### Project file & Debug log uploads

[rs-winder-all.3mf.zip](https://github.com/user-attachments/files/16514229/rs-winder-all.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

3D-Printguy, 2024-08-07:
This is a known bug. There is a workaround but no fix yet.

https://github.com/bambulab/BambuStudio/issues/3847

walterwongbbl, 2024-08-13:
@3D-Printguy @Erin-Mounts 
We currently do not support file transfer outside of LAN ONLY.

alejandrojrey, 2024-08-19:
> @3D-Printguy @Erin-Mounts We currently do not support file transfer outside of LAN ONLY.

Weel, then you should disable the option in the slicer

Erin-Mounts, 2024-08-20:
> @3D-Printguy @Erin-Mounts 
> 
> We currently do not support file transfer outside of LAN ONLY.

I see. Out of curiosity, when did you stop supporting this, and why? I'd been relying on it for well over a year before it stopped working. If it's not a bug, I'd like to make a feature request that you bring back this support.

rogocal, 2024-08-27:
> I see. Out of curiosity, when did you stop supporting this, and why? I'd been relying on it for well over a year before it stopped working. If it's not a bug, I'd like to make a feature request that [...]

Second this, this was one of the features that made me buy a Bambu printer. 
Taking this down without notice feels wrong and would like to see the functionality back on the SW. 

Besides, the main functionality is there (you can `Print plate` and the file is transferred over the internet but you can't just `Send` without printing)

walterwongbbl, 2024-10-17:
@Erin-Mounts @rogocal 
My statement is incorrect. We will use a new method to transfer files because there are too many issues with FTP. And the new transmission method supports all modes.