bambulab/BambuStudio #4596: Device Tab Filament section blinks frequently during print

bahamut44, 2024-08-07:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 and 11

### Printer

Bambu A1

### How to reproduce

During a print, while on the device tab, the filament section as well as the extruder blinks/flashes/auto refreshes itself. Seems to be at random intervals. 
I've tried rolling back two versions of Bambu Studio, with no change. I have a friend with an X1C and his studio does the same thing on version 1.9.3.50 as well. When watching this page while printing the flash/blink is very distracting. I also tend to stream my prints via twitch and this is a useful section that I show. It is very distracting for my viewers. This happens with any print. Sometimes it's infrequent but other times it happens every few seconds. It does not seem to corrilate with when the AMS is changing colors either. I can't really see a pattern to it, or a setting to toggle for it.

### Actual results

Example: https://jmp.sh/bncEhZJY

### Expected results

It would be nice if these sections of the program did not blink.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16517021/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bahamut44, 2024-09-10:
Anything?

walterwongbbl, 2024-09-18:
Hi.  @bahamut44  What is your computer screen refresh rate?
If you are not currently connected to AMS, will the page still flash.

bahamut44, 2024-09-20:
Refresh rate is standard 60 hz. Why does it flash at all? seems like a very distracting bug to me.