bambulab/BambuStudio #4598: Changing plates opens empty gcode files

bolsoncerrado, 2024-08-07:
### Bambu Studio Version

Any

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma 14.5

### Additional system information

MacStudio M1

### Printer

X1C, P1P

### How to reproduce

1. Open or create new project
2. Assign a TEXTURED plate
3. Close Bstudio
4. Try to open ANY gcode that previously was saved as COOL PLATE (for example)

### Actual results

The opened .gcode.3mf appears empty, you can't print or do anything with it.

### Expected results

The ideal would be no matter which plate was last set on the slicer, the .gcode should open anyway and you should be able to send the job to any of the printers.

The ONLY workaround is to OPEN Bstudio again, assign the correct plate you suspect the .gcode you want to open was set to and THEN open the .gcode.

Then it works :/

### Project file & Debug log uploads

Not needed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-08-08:
Hello, may I ask if you are using the latest version? I tried to reproduce it, but I couldn't. Can you record a software version and a GIF of the operation? Thank you

bolsoncerrado, 2024-08-08:
Yes, here you go:
https://www.dropbox.com/scl/fi/71m5lhfxhryrxiwgrkyxd/BSTUDIO-BUG.mp4?rlkey=c9ockc2ditj87299k670kqxmo&dl=0

bolsoncerrado, 2024-10-02:
https://github.com/user-attachments/assets/0af595aa-3faf-41de-8f4e-4ed1de0b3d33

bolsoncerrado, 2024-10-02:
@Haidiye00 reuploaded

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

Haidiye00, 2025-04-16:
Hello, please use the latest version, thank you.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

closed 2025-04-16 by Haidiye00

bolsoncerrado, 2025-04-16:
@Haidiye00  it still FAILS exactly the same way as it was on 1.9.xxx, using 2.0.2.57

Haidiye00, 2025-04-16:
Hello, could you please provide your 3mf? Thank you

bolsoncerrado, 2025-04-16:
As you can see in the videos it happens with any .gcode.3mf but let me try with one created WITH 2.xxx and see if it's just with gcode from 1.9.xxx

bolsoncerrado, 2025-04-16:
Afterwards I've modified some things on the .3MF, saved it with Bstudio 2.xxx, sliced it, saved the gcode with Bstudio 2.xx and quit Bstudio 2.x

Now I open Bstudio 2.xx
Create New Project
Switch plate to TEXTURED
Close BStudio 2.xx
Then I doubleclick the .gcode I just saved with Bstudio 2.xx and when loading you can clearly see it temporarily switches from Textured to Cool and Back to Textured, and the contents remain EMPTY.

And here comes the fun part!
Close BStudio 2.xx
Open Bstudio 2.xx
Create New Project.
PLate is STILL TEXTURED PLATE.
IF I DRAG AND DROP THE GCODE FILE INSIDE THE WORKING AREA, COOL PLATE IS LOADED AND SLICED FILE SHOWS FINE!!!!!

bolsoncerrado, 2025-04-16:
So yes, it happens on BOTH 1.9.xx and 2.xx but only if I DOUBLECLICK the gcode, not if I drag and drop it.

bolsoncerrado, 2025-04-16:
Heres the video, check how the plate changes when I double click vs when I drag n drop.

https://github.com/user-attachments/assets/61842238-0d6e-4756-99ae-0814ca977020

Haidiye00, 2025-04-17:
hello ,can you provide you gcode file?thanks

Haidiye00, 2025-04-17:
Hello, my email is zhou.xu@bambulab.com

bolsoncerrado, 2025-04-17:
I emailed you the file.

Haidiye00, 2025-04-18:
[image]
Suggest turning off this and trying it out,
May I ask how many printer devices you have?

bolsoncerrado, 2025-04-18:
A whole farm multidevoce can't handle :(