bambulab/BambuStudio #4601: Does not recognize Bambu ABS-GF filament

houlster, 2024-08-07:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambu X1

### How to reproduce

Load ABS-GF into AMS
Click Filaments in BS
ABS-GF does not show up in the AMS filament list.  The slot just doesn't show up.

### Actual results

ABS-GF does not show up

### Expected results

ABS-GF should show up

### Project file & Debug log uploads

There's nothing to attach

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2024-08-08:
@houlster Have you updated the profile？ When bambu studio boot up, there is a pop-up notification windows to ask user to update profiles.
[image]

houlster, 2024-08-10:
Hmm, no.  I haven't gotten any kind of prompt,  unless I just missed it.  I'll see if I can get it.In the mean time, I can still pick abs-gf to print, I just need to pick the AMS slot when I send to printer.

image.png (view on web)

QingZhangBambu, 2024-08-12:
@DanBao-Bambu please check this. @houlster please upload the log files to help us target the bug