bambulab/BambuStudio #4606: 4 AMS Can't add external filaments

Speakeriii, 2024-08-08:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Ryzen 7700x,, 64GB Ram DDR5 6000, RTX 4090, M.2 PCIE 4 SSD

### Printer

X1-Carbon

### How to reproduce

I added 4 AMS units to my printer.  (An update happened to be sent out the same day I added my printers) the Add a filament button no longer works for me.  I know I have 16 to choose from why do I need to add more.  I want to use TPU, I would really like to do multi colour TPU.  One saving grace is when I click resync it adds the external spool I had chosen before the update & upgrade.  Unfortunately when I switch the type it doesn't follow through to the Device tab.  I have to also switch the type of filament on the device tab for it to print using TPU settings.  I also am not able to add a 2nd spool so I can do multi-colour or type so no supports etc.

### Actual results

NOTHING when I click the + key nothing happens

### Expected results

The ability to add a filament I can select to do print job.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16553105/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ttt888zzz, 2024-08-09:
HI,[Speakeriii](https://github.com/Speakeriii):
Could you please provide a gif or a video，we will check it.

bambu123, 2024-08-09:
Currently, you can add up to 16 materials. Do you mean you need to add more materials, such as 20, to solve your problem?

Speakeriii, 2024-08-09:
Yes, if the max is set to 16, I can't use any material not supported by the AMS.  I know it's not a problem 99% of your customers have.  Sorry for being a pain in the butt.  If you could increase the limit in the next update that would be great.  I will unplug one of my AMS when I need to print with unsupported filament until you have a chance to add it to an update.Thank you for a quick response and giving me a way to work around the issue.Sent from my Galaxy