bambulab/BambuStudio #4608: Confusing "Stopped" message

GuzziRaz, 2024-08-09:
### Bambu Studio Version

latest

### Where is the application from?

Bambu Lab github releases

### OS version

n/a

### Additional system information

n/a

### Printer

X1C

### How to reproduce

While printing, turn video on then off. Now it says "Stopped"

### Actual results

When going to device tab from some other view, I'm always fooled by that "Stopped" message, thinking for a while that the **print** has stopped.

### Expected results

It should say "Video stopped" or "Video paused".

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-08-12:
HI，[GuzziRaz](https://github.com/GuzziRaz)：
Thank you for your advice.Do you mean that the“ stop ”here should be changed to “ video stop” 
[image]

GuzziRaz, 2024-08-12:
@ttt888zzz yes. ”Video stopped”.

ttt888zzz, 2024-08-13:
We will fix it in the next version, thank you for your thoughtful suggestions