bambulab/BambuStudio #4609: send to printer error from appimage

threepoint2, 2024-08-09:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

EndeavourOS x86_64, Linux 6.10.3-arch1-2

### Additional system information

Linux 6.10.3-arch1-2
Kernel: Linux 6.10.3-arch1-2
DE: KDE Plasma 6.1.4

### Printer

BambuLab X1C, 01.08.00.00

### How to reproduce

1. Slice file
2. Click on preview
3. Change from Print Plate to either Send or Send All (error occurs on both)
4. Either rename the file or just click send (error occurs either way).

### Actual results

Dialogue box titled "LAN Connection Failed (Sending print file)" with steps to find the LAN IP and Access Code of the printer.
The printer is not configured for LAN Only; it is configured for cloud printing and printing via the cloud works with no error.
See screenshot for error.
[image]

### Expected results

The file should be sent to the printer for printing via the printers display or remotely via the BambuHandy Application. 

### Project file & Debug log uploads

[debug_Fri_Aug_09_19_35_23_25227.log.txt](https://github.com/user-attachments/files/16565741/debug_Fri_Aug_09_19_35_23_25227.log.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

threepoint2, 2024-08-10:
I have been able to replicate the same issue on a windows virtual machine where the network adapter is set to NAT and not Bridged. I suspect the issue in the Appimage is network related. Since there is no way to configure networking in the Appimage, could this be they way it's configured?

MackBambu, 2024-08-13:
The send file can only be used on the LAN, not through the cloud. So if your pc and printer are not on the same LAN, this dialogue  will pop up.

threepoint2, 2024-08-13:
I know it can only be used on the LAN. My computer and the printer are on the same LAN however when I choose send to printer it pops up the issue; only on the fedora appimage on an Arch based machine. The unbuntu appimage and windows clients both work as expected. This is a bug.

threepoint2, 2024-08-13:
Edited previous comment. This is not a bug after all. The default firewalld service on EndeavourOS was blocking comms; turning the firewall off solves the problem. Any idea what ports the printer communicates on over the LAN?

threepoint2, 2024-08-13:
I fixed the issue:
firewalld log files showed

> Aug 13 16:59:00 blasterdroid-desktop kernel: filter_IN_home_REJECT: IN=enp2s0 OUT= MAC=70:85:c2:d5:57:70:08:fb:ea:43:9e:26:08:00 SRC=192.168.1.24 DST=192.168.1.21 LEN=456 TOS=0x00 PREC=0x00 TTL=64 ID=2287 DF PROTO=UDP SPT=2021 DPT=40886 LEN=436

fixed by adding the right port

`sudo firewall-cmd --zone=home --add-port=2021/udp --permanent`

threepoint2, 2024-08-13:
Issue closed. See comments for fix

closed 2024-08-13 by threepoint2