bambulab/BambuStudio #4610: "选择性应用斜拼接缝"在特定细小圆柱体上出现错误的切片结果并导致打印成品低表面质量

moe001, 2024-08-09:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Printer

p1sc

### How to reproduce

1. 载入附带工程
2. 切片并观察预览结果

### Actual results

其中一个圆柱体的表面不规整，似乎斜拼只应用在了部分层。这两个圆柱体唯一的差别就是是否开启了题述选项，并且原模型建模均匀，面数充足，不应该有此情况。
图片中可见左侧圆柱体右半部分的异常。
[image]
这个问题也会反应到加工成品中，下述示例图中可见不同层的接缝模式不同。（涂抹了墨水以凸显差异）
[image]

### Expected results

更加精确的预测是否应用斜拼，至少在这种（预期效用中）每层一致的模型上，不应该随着层数出现随机的结果。

### Project file & Debug log uploads

[bug240810.zip](https://github.com/user-attachments/files/16566081/bug240810.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-08-13:
@moe001 你好，这个问题我已经知道，是因为角度检测不够准确，这个角度检测后续我会默认关闭，保证除了悬垂部分全部应用。您在使用时可以将角度阈值调整到0。

QingZhangBambu, 2024-10-12:
@moe001 测试了多个模型之后，智能应用还是很难涵盖所有模型。目前暂时还是需要您手动调整。后续会继续努力优化

closed 2025-05-23 by QingZhangBambu