bambulab/BambuStudio #4612: Crash when changing speed to Sport mode

theosib, 2024-08-10:
### Bambu Studio Version

1.

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.5

### Additional system information

MacBook Pro M2 Pro

### Printer

X1C

### How to reproduce

Send print job
Change speed to sport

### Actual results

Crashed

### Expected results

Should not crash

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16568368/log.zip)
[ams riser 3-left front combo002.3mf.zip](https://github.com/user-attachments/files/16568369/ams.riser.3-left.front.combo002.3mf.zip)
[bambu crash 2.txt.zip](https://github.com/user-attachments/files/16568370/bambu.crash.2.txt.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

theosib, 2024-08-10:
Bambu Studio has been crashing a lot lately, so this bug could be a duplicate of https://github.com/bambulab/BambuStudio/issues/4600, although I'm not sure. This time it didn't crash when sending the job to the printer but rather when I tried to change the print speed.

iCMDdev, 2024-08-10:
Out of curiosity, I've looked at both (this issue and the other one you mentioned) your crash reports generated by macOS, as well as other apps' similar crash reports I found on the web, and it seems related to [wxWidgets](https://github.com/bambulab/wxWidgets) (and GUI stuff) used by Bambu Studio. 

This particular issue seems to finally end up crashing [somewhere in this function](https://github.com/bambulab/wxWidgets/blob/66a0ce3c438e1807a14bf5065a8606557d3193c7/src/osx/cocoa/nonownedwnd.mm#L122 at one of the pointer dereferences. My **guess** is that this could be a GUI race condition bug.

walterwongbbl, 2024-08-27:
@iCMDdev 
Actually, we haven't made any changes here. We found that M2 Mac computers seem to be more prone to such problems.

iCMDdev, 2024-08-27:
@walterwongbbl 
> Actually, we haven't made any changes here.

Perhaps it is a problem within wxWidgets as you suggested in the other issue's thread.

> We found that M2 Mac computers seem to be more prone to such problems.

I recently experienced a similar crash on an M1 Pro device (#4696), though the device was on a Developer Beta build of macOS Sequoia, so that could've caused it.

simonwex, 2025-03-06:
Same crash on Macbook M3 pro.