bambulab/BambuStudio #4614: Unable to Connect to Printer in LAN only mode

scottpreston, 2024-08-10:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Sonoma

### Printer

X1 Carbon

### How to reproduce

Add Printer
Enter Code
System either doesn't connect or bambu studio crashes
[image]
[image]

### Actual results

Crash or Eror

### Expected results

Connects to printer so i can print.

### Project file & Debug log uploads

No file. No Project.

### Checklist of files to include

- [X] Log file
- [X] Project file

scottpreston, 2024-08-10:
was able to do a factory reset and it seemed to work

closed 2024-08-10 by scottpreston

MarkJeronimus, 2024-09-17:
I experienced the same with BS 1.9.5. I'm not in the mood for doing a factory reset because I'm printing a 2 minute print that I really need, and downgrading to 1.8.4 worked like a charm.

<sub>P1P with FW 1.2.2.6, used in LAN-only mode</sub>