bambulab/BambuStudio #4615: Bambu Studio closes when trying to slice model

ChukkarNaut, 2024-08-10:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Slice plate

### Actual results

Application closes

### Expected results

Slice and ready to print

### Project file & Debug log uploads

[debug_network_Sat_Aug_10_12_54_26.log.zip](https://github.com/user-attachments/files/16571111/debug_network_Sat_Aug_10_12_54_26.log.zip)
[Zelda_pot-180pct.zip](https://github.com/user-attachments/files/16571113/Zelda_pot-180pct.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-08-12:
Could you please try the following steps, it may help：
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

zhimin-zeng-bambulab, 2024-08-26:
It is a bug, we will try to fix it. Thanks for reporting this issue!