bambulab/BambuStudio #4618: Network Plug-in for Raspberry Pi 5 aarch64-systems

lcrowser, 2024-08-11:
I have successfully built Bambu Studio on a Raspberry Pi 5 with 8GB of RAM .  Everything appears to be working with the exception of the Network Plugin.  In issue #3632, Network plugin is downloaded but not loaded the issue is discussed.  Can this plugin be built and made available to those who are using the P1 5?

BobbyMcJoe, 2024-08-18:
Just out of curiosity, did you install via Flatpak or Pi-Apps? I got the same issue in both install methods.

lcrowser, 2024-08-18:
I tried the install via Pi-Apps which uses Flatpak and also built an AppImage from the source code and have the same issue. 

Per this post,  https://github.com/bambulab/BambuStudio/issues/3632 the issues is known but in the post there appears to be no plan to have it available for aarch64-systems.  

I did add a feature request asking to have the plugin available for aarch64-systems.

BrettRogersUK, 2024-09-16:
I have this same problem. I'm using pi400 and have installed via pi-apps (flatpak). It says install successfully yet still says not detected.

I have read previous post #3632 and have also seen that they have no plans for it on aarch64 systems.

I see no point in developing the software for various platforms like aarch64 if you're not gonna develop the other bits that go with it. It doesn't make any sense IMHO.

yellobello, 2024-09-21:
Hi! Is the binary of the network plugin a x64 or x86 type? if it´s x86 there *might* be a chance to run it on wine, possibly using hangover. if x64 the chances are a lot slimmer. Also i´m absolutely uncertain if the windows plugin even wants to talk to the linux app, possibly the whole OS layer of inter-process communication is not there.
Maybe someone more knowledgeable could answer these questions before time is wasted on trying.
Can anyone confirm the binary type, if x86 I would volunteer to give it a try

meladori29, 2024-10-18:
I just got a Raspberry Pi 5 and also wondering if this will be fixed?  It was the entire point to buy a cheap PC to use for my Bambu printers.

dannutu, 2024-10-19:
> I did add a feature request asking to have the plugin available for aarch64-systems.

Where? I'd like to add a vote there, although I am almost sure it won't change anything

lcrowser, 2024-10-19:
It was on the original post thread that you viewed.  I don’t believe that there is any way to vote.  You might want to just create another feature request

jthomeo, 2024-10-24:
Another one on the same boat. Bambu studio without the network plugin is not as useful as it should be.

dannutu, 2024-11-15:
I _guess_ they just can't be bothered to build it for Linux on ARM. I can't see why it would not be possible.

BlackDragon-B, 2024-12-18:
Same issue here. Running a postmarketOS chromebook with BambuStudio installed as a flatpak.

trvsm, 2025-01-02:
I've got this issue as well, running on 64 bit raspbian

M1chlCZ, 2025-04-28:
Still the same issue

marekedg, 2025-09-19:
So has anyone found a fix yet I have the same issue I think its because It doesnt inherently support it on arm based chips but i may be wrong