bambulab/BambuStudio #4619: Cannot use fonts with the text tool.

Keri010, 2024-08-11:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sonoma 15.5

### Printer

Bambu A1, 

### How to reproduce

Add text, then change the font using the drop down menu.  [
[Screen Recording 2024-08-11 at 3.56.59 PM.zip](https://github.com/user-attachments/files/16575956/Screen.Recording.2024-08-11.at.3.56.59.PM.zip)
](url)

### Actual results

Some of the fonts change, but many fonts do nothing.   

### Expected results

The font of the text should change to the selected font.  

### Project file & Debug log uploads

[Hulk updatefont.3mf.zip](https://github.com/user-attachments/files/16575941/Hulk.updatefont.3mf.zip)
[Archive.zip](https://github.com/user-attachments/files/16575970/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-08-12:
Known issue, thanks for the feedback

adamurban, 2025-01-04:
Can you please link to the bug you are tracking this with and advise when we can expect a fix? This is still an issue in the latest software, and this is a pretty important feature...

```
Font_FontMgr, warning: unable to find font 'Apple Chancery' [bold]; 'Arial' [aspects: regular,bold,italic,bold-italic] [paths: /System/Library/Fonts/Supplemental/Arial.ttf;/System/Library/Fonts/Supplemental/Arial Bold.ttf;/System/Library/Fonts/Supplemental/Arial Italic.ttf;/System/Library/Fonts/Supplemental/Arial Bold Italic.ttf] is used instead
```

Haidiye00, 2025-01-06:
Hello, we are currently working on resolving this issue