bambulab/BambuStudio #4620: Bambu Studio crashes when slicing a model

j-collom, 2024-08-11:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS Sonoma 14.5

### Additional system information

CPU: 2.3 Ghz 8-core Intel core I-9
Graphics: Radeon Pro 560X 4 GB
Intel UHD Graphics 630 1536 MB
Memory: 16 GB 2400 MHz DDR4

### Printer

Bambu Labs P1S

### How to reproduce

Slice a model and the application shutsdown with the following error - MAC OS SIGNAL, Code 11 Segmentation fault: 11

### Actual results

The application hals and closes

### Expected results

The slicer should complete the slicing and allow me to print the item

### Project file & Debug log uploads

[BambuStudio_Logs.zip](https://github.com/user-attachments/files/16576020/BambuStudio_Logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-08-13:
@j-collom  Is the crash caused by a special file? Could you share your 3mf file?

j-collom, 2024-08-14:
[Bambu_Studio_memory_leak_files.zip](https://github.com/user-attachments/files/16617095/Bambu_Studio_memory_leak_files.zip)

It appears there are 2 problems which lead to toolpath creation causing a memory leak, these files together cause the slicer to hang forever until it crashes. The other problem is when your project file gets too complex (lots of different plates and models on the plates) when you slice a plate with a model or models it just crashes immediately. What I did was load the entire 'Voron V0.2' build as a project, using different plates. As the project got more complex it just crashed every time I tried to slice.. It seems both problems are memory issues as the crash was a SIGSEGV Signal 11 so something tried to stomp on a used memory segment and the problem with toolpath used up all the memory on my computer and it started throwing memory alerts over and over even after closing all other apps, leaving just Bambu Studio open and trying to slice it still used up all my systems memory and ultimately crashed.

zhimin-zeng-bambulab, 2024-08-26:
@j-collom 
Could you please share the 3mf file that caused the crash?
The zip package you sent contains stl files. I have tried many times but the crash problem has not been reproduced.

zhimin-zeng-bambulab, 2024-10-21:
@j-collom  Could you reproduce this problem in 1.10 Beta 1?

j-collom, 2024-12-30:
No issue resolved

closed 2024-12-30 by j-collom