bambulab/BambuStudio #4624: 100% repeatable (4 times out of 4) nozzle blockage on a print.

ChudsterGit, 2024-08-12:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1 Carbon

### How to reproduce

Import and slice model
Send to printer

### Actual results

Around layer 45 the printer stops feeding filament through the nozzle. No error is reported and the printer continues regardless, not printing.

### Expected results

Upon stopping the print there is a 'cannot retract filament' error. At this point the entire printer head/extruder assembly has had to be disassembled, and the stuck filament removed. This has happened 4 times in a row.

Expected
[image]

Point of failure
[image]

Results
[image]
[image]

In the final attempt I realised there was an error seeming caused by the file itself, so changed the infill (just to make a change to the model file) but the result was the same.

### Project file & Debug log uploads

Project File
[Short Edge Test.zip](https://github.com/user-attachments/files/16590977/Short.Edge.Test.zip)

Log File
[debug_Sat_Aug_10_14_44_59_22560.log.zip](https://github.com/user-attachments/files/16590991/debug_Sat_Aug_10_14_44_59_22560.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-08-13:
HI [ChudsterGit](https://github.com/ChudsterGit),
Is there a red exception message here, like the picture below
[image]

ChudsterGit, 2024-08-13:
Hi there - sorry, no - no red box. The print continues as if there is filament passing through the extruder, but nothing comes out. In each attempt I have manually stopped the print (the first time at 99% 'complete') - and it is only after stopping it that I received the message about not being able to retract the filament. This is accompanied by a red flashing light on the AMS. 

From then on I have had to completely disassemble the extruder each time. In my 10 months of owning the printer I have only had 2 other filament jams - it is strange that I get 4 now in exactly the same place on the same print. 

When trying to un-jam the filament I can manually 'rock' the yellow drive wheel, about an 8th of a turn before it tightens too much, and can see the filament move a little. But these are the extents of home much the filament can move - maybe 2-3mm worth.

ChudsterGit, 2024-08-14:
Hi there - in addendum to this issue, I printed two of the models simultaneously and they both completed fine.