bambulab/BambuStudio #4626: Linux workspace rendering issues

St0nefish, 2024-08-13:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Nobara 40 (Fedora 40)

### Additional system information

CPU: Ryzen 5800X
GPU: NVIDIA RTX 3080TI

### Printer

Bambu Lab P1S

### How to reproduce

1. Install Bambu Studio via the Fedora AppImage *or* Flatpak (issue happens with both)
2. Launch Bambu Studio

### Actual results

Setup works, printer config works, camera preview works, workspace does not. I can import STL files (via File > Import menu), I can even see the right click menu on the "workspace" and add primitives, but nothing renders, including the menu across the top. 
I see identical results launching the AppImage *or* installing and launching via Flatpak (tried this after seeing a suggestion in a reddit post - it is easier to keep updated this way)

### Expected results

workspace should render the build plate and the imported models along with a toolbar across the top

[image]

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/16592859/logs.zip)

skipped project file - reproducible without actually starting any projects

### Checklist of files to include

- [X] Log file
- [ ] Project file

MackBambu, 2024-08-13:
```
got opengl version 4.6.0 NVIDIA 560.31.02, glsl version 4.60 NVIDIA, vendor NVIDIA Corporation
```
Can you test the mesa driver, usually there is no problem with the mesa driver.

St0nefish, 2024-08-13:
> GPU: RTX 3080TI

considering that this PC is primarily for gaming and that I'd be massively crippling an extremely high-powered GPU just to get a slicer working, I'm going to veto that suggestion. I'd be better off setting up a VM or trying the windows version via Proton/Wine

however, I question why you bother having a linux native version if you can't support, or apparently haven't even tried, the proprietary NVIDIA drivers when the vast majority of consumer GPUs are NVIDIA. I would also point out that this is the only application I'm seeing this issue with, I've run a ton of other software entirely with the NVIDIA drivers without any issues.

St0nefish, 2024-08-14:
update: Proton/Wine was rendering the workspace, but was having other issues (startup wizard was all white, home page was blank, unable to log in) so I ended up just creating a Windows 10 VM via QEMU/KVM and that does work without forcing me to actually install windows. 

however, while I do have a workaround, this *is* still a bug, and I will continue to monitor it. the native version should be capable of running with the NVIDIA drivers as those are essentially required to get full functionality out of their cards and I would much rather use the linux version and avoid having to boot a VM just to run the slicer. 

fwiw I also tried the Orca Slicer linux build but it has the exact same issues, likely coming from the upstream Bambu version, but I figured it was worth a shot.

MackBambu, 2024-08-27:
Downgrading to nvidia 555 is ok?
https://github.com/bambulab/BambuStudio/issues/4669

tobydeh, 2024-08-30:
@MackBambu I can confirm downgrading to 555 fixes the problem.

oryschakj, 2024-08-31:
> update: Proton/Wine was rendering the workspace, but was having other issues (startup wizard was all white, home page was blank, unable to log in) so I ended up just creating a Windows 10 VM via [...]

These environment variables restore the rendering functionality in Orca Slicer: 

`__GLX_VENDOR_LIBRARY_NAME=mesa
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json`

https://wiki.archlinux.org/title/NVIDIA/Troubleshooting

I tested them with Bambu Slicer but didn't have the same luck. It's probably an issue with the Flatpak environment.

Temporary workaround, obviously.

St0nefish, 2024-09-01:
I can confirm what oryschakj posted above, I actually found it last week and entirely forgot to update this issue. 

I'm using the AppImage which I manage via AppImageLauncher. In the config for Bambu Studio and Orca Slicer on the Application tab I have the following environment variables set: 
`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1`

with these configured both slicers are working (mostly) perfectly. opening the camera stream to Orca causes it to crash every time, but Bambu Studio is working without any issues at all that I've noticed. 

however, these should either be set in the AppImage or fixed to work with the latest NVIDIA drivers, so I'm going to leave the issue open for now

masakk1, 2024-09-29:
> I can confirm what oryschakj posted above, I actually found it last week and entirely forgot to update this issue.
> 
> I'm using the AppImage which I manage via AppImageLauncher. In the config for Bambu Studio and Orca Slicer on the Application tab I have the following environment variables set:
> 
> `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1`
> 
> I found that with only the VENDOR_LIBRARY settings the workspace functioned but the homepage did not. adding the COMPOSITING_MODE flags resulted in everything working as expected.
> 
> with these configured both slicers are working (mostly) perfectly. opening the camera stream to Orca causes it to crash every time, but Bambu Studio is working without any issues at all that I've noticed.
> 
> these should either be set in the AppImage or fixed to work with the latest NVIDIA drivers, so I'm going to leave the issue open for now

This didn't really work for me
[image]

and neither did the previous suggestion.

I am using a NVIDIA 3060 with the drivers 560
Using Nobara 40 (Fedora)

Theses are some of the errors that i get when opening the "Prepare" tab
```
(bambu-studio:6987): GLib-GObject-CRITICAL **: 13:25:00.888: invalid cast from 'wxPizza' to 'GtkCellLayout'
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:00.888: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:01.317: gtk_window_resize: assertion 'height > 0' failed
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:01.317: gtk_widget_set_size_request: assertion 'width >= -1' failed
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:01.317: gtk_widget_set_size_request: assertion 'width >= -1' failed
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:01.318: gtk_window_resize: assertion 'height > 0' failed
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:01.320: gtk_window_resize: assertion 'height > 0' failed
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:01.321: gtk_window_resize: assertion 'height > 0' failed
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:01.321: gtk_widget_set_size_request: assertion 'width >= -1' failed
(bambu-studio:6987): Gtk-CRITICAL **: 13:25:01.321: gtk_widget_set_size_request: assertion 'width >= -1' failed
```

masakk1, 2024-09-29:
after some more testing:

1. Prusa slicer AppImage opens and seems to work
2. Bambu studio AppImage has a very clear issue where the workspace doesn't even render.
3. Orca Slicer AppImage is not even rendering the text - however, when i try to drag text, i can see that there was text - i could'nt get farther than the hello screen since it was all blank

TyskieGronie, 2024-10-03:
Same problem with Pop!_OS 22.04 since updating from nvidia 555 to 560.
If start a X11 session it works. On wayland no building plate. Prusa Slicer works.

A solution would be greatly appreciated.

johnquix, 2024-10-08:
So is it confirmed that this is a Wayland issue? I ask because most of my other devices are using Wayland but this is my only device using Wayland with Nvidia. I am assuming that some portion of this has to do with Nvidia drivers not being loaded correctly when using Bambu Studio on a Wayland + Nvidia combination. 

Is anyone having this issue in a different configuration than above?

joshuasing, 2024-11-04:
I am also experiencing this issue on Ubuntu 24.10 x86_64 (Linux 6.11.0-9-generic), with Wayland. I use a  NVIDIA GeForce RTX 3060 with proprietary driver version 560.35.03. Uninstalling/reinstalling (including deleting cache/config/other related files) does not help.

[image]

TerraTreader, 2024-11-15:
I am also having this issue. Fedora 41 / NVIDIA 565 (cannot downgrade), RTX 2080. Attempted to add the Environmental Variables into the Flatpak permissions. That did not work. Added the variables into AppImage through AppImageLauncher and that resolved the issue. 

Still feel this a bug and these options are just workarounds. Wasn't needed in Fed 40/NVIDIA 555 drives. Would be good to have development resolve the issue

DeeKahy, 2024-11-19:
Switching to X11 from wayland solves the issue whilst using NVIDIA 560. I haven't tested on 565 yet.

powermad80, 2024-12-28:
Happening to me too, still, on Fedora 41 KDE. Adding the environment variables works for getting the build plate preview to display but then I get IO errors when attempting to actually load a model in a project, so printing is impossible. At least the flatpak lets me print even though I can't get the preview working with it, even when setting it to run in X11 as others have.

mmayla, 2025-01-06:
I can confirm that it's a Wayland issue. When I switch to X11 everything works.

jdecoste, 2025-01-06:
Can confirm this is an issue for me as well. Fedora 41, 3080ti, driver version 565.77, Wayland

RickBankers, 2025-01-10:
Are there any fixes for this with the flatpak version? I have to use the flatpak version.

igl-x2, 2025-01-15:
> This fixed the flatpak version for me:
> 
> Through the `flatseal` app disable `All devices (e.g. webcam)`

this is no fix as it reverts back to software rendering (like all other 'solutions')

YoSTEALTH, 2025-01-18:
Can't use Bambu Studio under Wayland session. This is very annoying, everything else (OpenScad, ...) works.

Please fix this ASAP, Thanks.

``` bash
❯ bambu-studio
[2025-01-18 12:42:02.711558] [0x00007ec97884b240] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/<user>/.config/BambuStudio/plugins/libbambu_networking.so
after dlopen, network_module is 0x145374d0

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:02.835: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.835: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:02.835: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.835: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:02.837: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.837: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:02.837: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.838: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:02.839: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.839: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:02.840: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.840: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (bambu-studio:157211): CRITICAL **: 12:42:02.906: Cannot register URI scheme wxfs more than once

** (bambu-studio:157211): CRITICAL **: 12:42:02.906: Cannot register URI scheme memory more than once

** (bambu-studio:157211): CRITICAL **: 12:42:02.907: Cannot register URI scheme wxfs more than once

** (bambu-studio:157211): CRITICAL **: 12:42:02.907: Cannot register URI scheme memory more than once

** (bambu-studio:157211): CRITICAL **: 12:42:02.908: Cannot register URI scheme wxfs more than once

** (bambu-studio:157211): CRITICAL **: 12:42:02.908: Cannot register URI scheme memory more than once

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.917: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.918: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.918: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.918: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.918: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.918: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.918: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.918: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.918: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.921: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:02.921: gtk_widget_set_size_request: assertion 'width >= -1' failed

[... 512 lines cut ...]
(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:29.368: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:29.368: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:29.372: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:29.372: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:29.375: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:29.375: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:29.377: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:29.377: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:29.380: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:29.380: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:157211): GLib-GObject-CRITICAL **: 12:42:29.380: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:157211): Gtk-CRITICAL **: 12:42:29.380: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

```

ajunca, 2025-02-20:
Software rendering is very slow, and for "big" models barely usable... I found this parameters using zink (I think founded in orca slicer issue thread) that works for me using hardware acceleration (nvidia user):
`
__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink WEBKIT_DISABLE_DMABUF_RENDERER=1 bambu-studio`

St0nefish, 2025-02-20:
I'd been looking to build a new rig anyway and after seeing both the insane pricing and borderline paper launch of the 50 series decided to try an AMD GPU instead, so picked up a 7900 XTX. 

It's been a *vastly* better linux experience and "fixed" this issue, so further confirms that the problem is Wayland + nvidia

This probably isn't a solution for most people, but I figured it's worth mentioning for anyone considering a new GPU anyway.

powermad80, 2025-02-21:
> I found this parameters using zink (I think from orca slicer issue thread) that works for me using hardware acceleration (nvidia user):
> 
> ```
> __GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink WEBKIT_DISABLE_DMABUF_RENDERER=1 bambu-studio
> ```

Tried it on OrcaSlicer, this works for me! It's still a bit choppy to do things like window resize operations but build plate operations are smooth as butter now! I can rotate the build plate smoothly even after slicing now, thanks!

powermad80, 2025-02-22:
Addendum: that set of parameters somehow *doesn't* work on BambuStudio, whether the flatpak or appimage. At most I just get the build plate visible but horribly choppy. No idea why this would work so cleanly on Orca and not Bambu.

glutenmancy, 2025-02-22:
For:
- Bambu Studio 1.10.1 flatpak from flathub 
- Fedora 41
- Nvidia GTX 1080ti with Nvidia driver 565.77

These env vars work for me applied with flatseal:
- WEBKIT_DISABLE_DMABUF_RENDERER=1
- GALLIUM_DRIVER=zink
- MESA_LOADER_DRIVER_OVERRIDE=zink
- __GLX_VENDOR_LIBRARY_NAME=mesa
- __EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json

The preview window clearly works will hardware acceleration. No choppiness.

Hope Bambu Lab can resolve this issue soon for everyone - without overrides.

powermad80, 2025-02-22:
Reinstalled the 1.10.1 flatpak and double checked all those env variables, put them in exactly that order (can't imagine that matters but did it anyway), and now it works! I've got Bambu Studio and OrcaSlicer running mostly nice and smoothly.

joshuasing, 2025-02-24:
I was able to get both Bambu Studio and OrcaSlicer working on Ubuntu 24.10 using the AppImages with:
```
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
```

ajunca, 2025-02-24:
I think (if I'm not mistaken) with that approach @joshuasing  is that it is not using hardware rendering. I suggest to try the zink method and see if there is a perceptible difference in rendering sluggishness (better try with an sliced model, as software rendering starts to have problems with that).

joshuasing, 2025-02-25:
> I think (if I'm not mistaken) with that approach [[image]@joshuasing](https://github.com/joshuasing?rgh-link-date=2025-02-24T11%3A16%3A30.000Z) is that it is not using hardware rendering. I suggest to try the zink method and see if there is a perceptible difference in rendering "sluggishness/responsiveness" (better try with an sliced model, as software rendering starts to have problems with that).

Thanks. You might be right, it did feel slower than usual with just that environment variable.
I added the other variables listed here https://github.com/bambulab/BambuStudio/issues/4626#issuecomment-2676350419 and it is working and feels same as with X11.

publicvirtualvoid, 2025-02-28:
Wow, I hadn't checked in on this in a while but assumed it would have been resolved by now. Pretty embarrassing to have such a maker-oriented product not work properly on Linux for such a long time.

YoSTEALTH, 2025-02-28:
> Wow, I hadn't checked in on this in a while but assumed it would have been resolved by now. Pretty embarrassing to have such a maker-oriented product not work properly on Linux for such a long time.

@publicvirtualvoid Bambu Studio works just fine on Linux and x11 (which most people use). It just has miner issues with Wayland. There is no need to dis.

powermad80, 2025-02-28:
I would hardly call it a "miner" issue when the build plate and all its controls are straight up invisible without an environment variable hack fix, which still leaves it performing so atrociously that controls in the preview are essentially unusable! That's a major issue!

Also, x11 is not nearly as dominant as it once was. Sure plenty of people still stick to it, but every major distro packs in wayland by default. It is the default configuration for any new setup nowadays, whether through GNOME, KDE, Hyprland, Cosmic, or some other thing. Let's not downplay very real and serious issues!

YoSTEALTH, 2025-02-28:
If you want it that badly nothing is stopping you guys from creating a pull request. I would like to see this issue fixed as much as everyone here. I do get where you are coming from though, I just had to switch to using x11 myself, it really is that bad.

publicvirtualvoid, 2025-02-28:
@YoSTEALTH I don't think anyone here needs you to police what is acceptable service. This issue was acknowledged a long time ago and prevents a paid product from functioning correctly under normal operating conditions. So far we've just had dismissive responses from Bambu on this thread and the person assigned on their side is now inactive on GH altogether. It's likely that this has fallen through the cracks which makes expressing discontent entirely reasonable.

Gdetrane, 2025-03-23:
Here to confirm adding the env vars mentioned by @glutenmancy to flatseal works in general, however I noticed Bambu Studio becomes unresponsive while printing and trying to play the video feedback

I'm running Bambu Studio from flathub, arch linux, GNOME on Wayland, on an RTX 2070 S

NVIDIA Driver Version: 570.124.04     CUDA Version: 12.8

dyfrgi, 2025-03-28:
Thank you @glutenmancy! I can confirm that also works on NixOS, by using Flatseal to set those env vars - I assume it's using the files inside the flatpak which is why the LHS path to the config file works.

I am currently running bambu-studio 2.0 from flathub.

Sure would be nice to see this fixed.

m-m-adams, 2025-04-07:
That's working for me too, thanks!

dyslexicanaboko, 2025-04-22:
@glutenmancy dood I could hug you right now. Thank you so much for posting that tip about the Flatseal environment variables. Saved my bacon.

Just to add to the stack here. I am running an NVIDIA RTX 4070. I, like most, don't think `down grading drivers` is a good solution. Everything else works. I installed Prusa Slicer, works fine. I use Blender, Steam, FreeCAD all of these things work. No way is it a drivers issue. Here is the weirdest part. I haven't changed anything about my system at all and Bambu Studio stopped working properly all of a sudden last week out of the blue. I didn't update anything unless Flatpak did it on its own somehow?

To add insult to injury, I went ahead and installed it on a virtual machine running Windows 11. Then when I try to install the network driver the whole application bombs. Deleted the network driver DLL and it operates again 🤷🏻. So I was doing this really stupid thing where I was doing the slicing on a VM and passing it to my Linux box so I could send it to the printer. Just totally wacky.

I will keep poking around, but I really don't think it's a driver issue.

mdXcoding, 2025-04-24:
> For:
>
>     * Bambu Studio 1.10.1 flatpak from flathub
>
>     * Fedora 41
>
>     * Nvidia GTX 1080ti with Nvidia driver 565.77
>
>
> These env vars work for me applied with flatseal:
>
>     * [...]

thanks! this finally did it for me too, in exact order for:

Bambu Studio 2.0.2 flatpak 
Fedora 41
Nvidia RTX 3080 with driver 570.133.07

DemonHunter759, 2025-05-03:
Works on Bazzite 42
with Wayland
Bambu Studio 2.0.3 flatpak
Nvidia Driver 570.144

THANKS ALOT, now i can switch even more to Linux. Eventually i can stop dualbooting

ywaf, 2025-05-05:
Thank you for the fix! Applied with flatseal on Fedora 42 - rtx 3070 on driver 570.144. All is working :)

BobbyG333, 2025-05-12:
> For:
>
>     * Bambu Studio 1.10.1 flatpak from flathub
>
>     * Fedora 41
>
>     * Nvidia GTX 1080ti with Nvidia driver 565.77
>
>
> These env vars work for me applied with flatseal:
>
>     * [...]

This work for my on
# System Details Report
---

## Report details
- **Date generated:**                              2025-05-12 12:40:26

## Hardware Information:
- **Hardware Model:**                              ASUS ROG MAXIMUS XII FORMULA
- **Memory:**                                      64.0 GiB
- **Processor:**                                   Intel® Core™ i9-10900K × 20
- **Graphics:**                                    NVIDIA GeForce RTX™ 3060
- **Disk Capacity:**                               2.0 TB

## Software Information:
- **Firmware Version:**                            2301
- **OS Name:**                                     Ubuntu 24.10
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               47
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.11.0-25-generic

acesyde, 2025-05-26:
@mdXcoding it works thanks

loneregister, 2025-06-12:
I created a bash shell, that I run as a program with the following commands in the .sh file.

`export __GLX_VENDOR_LIBRARY_NAME=mesa
export __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
~/Apps/Bambu_Studio_linux_fedora-v02.01.00.59.AppImage`

VonDecoy, 2025-07-20:
> For:
>
>     * Bambu Studio 1.10.1 flatpak from flathub
>
>     * Fedora 41
>
>     * Nvidia GTX 1080ti with Nvidia driver 565.77
>
>
> These env vars work for me applied with flatseal:
>
>     * [...]

@glutenmancy You're a hero to me now. 
I'm running Nobara Linux 42, KDE Plasma 6.4.2, Wayland, with a 3090. You've just solved my problem. I copied it exactly as written.

Caitex212, 2025-07-30:
> This didn't fix the flatpak version but made it run well enough for me:
> 
> Through the `flatseal` app disable `All devices (e.g. webcam)`

Actually that helped, the other solutions above didn't fix it. Thanks man

erindru, 2025-08-15:
It stopped working for me as well. I rolled back to the previous version:

```
$ sudo flatpak update --commit 89ea13aa4a965cfd6a5538503f905b16edc24a416e94d88f68cef734e7504807 com.bambulab.BambuStudio
```

and it started working again

commit list fetched via:

```
$ flatpak remote-info --log flathub com.bambulab.BambuStudio
```

TerraTreader, 2025-08-21:
I found another resolution with the fedora appimages for the issue that came up with the latest updates of fedora. 

I was missing mesa-compat-libOSMesa-25.0.4-1.fc42 - I installed that and was good to go.

SolidGlenn, 2025-12-09:
I've made a desktop file that seems loading the latest PR-8834 under ubuntu 25.10 with wayland and Nvidia 580.95.05 driver

`
[Desktop Entry]

Name=Bambu Studio  
Exec=~/Downloads/Bambu_Studio_ubuntu-24.04_PR-8834.AppImage
Comment=
Exec=env WEBKIT_DISABLE_DMABUF_RENDERER=1 GALLIUM_DRIVER=zink WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 __GLX_VENDOR_LIBRARY_NAME=mesa  __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ~/Downloads/Bambu_Studio_ubuntu-24.04_PR-8834.AppImage %F
Terminal=false
PrefersNonDefaultGPU=true
#Icon=/home/path/BambuStudio.png
Type=Application
Hidden=false
NoDisplay=false
StartupNotify=false
Terminal=false
`