bambulab/BambuStudio #4627: Segmentation fault when using BambuStudio CLI

uSyzex, 2024-08-13:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 

### Additional system information

i7 6700k, 32GB RAM, 3070Ti

### Printer

X1C

### How to reproduce

1. Try to slice a 3MF file using the CLI
2. ` ./bambu-studio.exe  --allow-newer-file  --slice 01  model_test.3mf --debug 5`

### Actual results

Segmentation fault

### Expected results

3mf sliced

Slicing works when using .stl files but not with 3mf even with the option allowing newer files. 

### Project file & Debug log uploads

```
[2024-08-13 12:22:09.029552] [0x00007c28] [trace]   Initializing StaticPrintConfigs
[2024-08-13 12:22:09.043549] [0x00007c28] [warning] cli mode, Current BambuStudio Version 01.09.03.50
[2024-08-13 12:22:09.043549] [0x00007c28] [info]    Will start to read model file now, file count :1

[2024-08-13 12:22:09.043549] [0x00007c28] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0, skip_useless_pick 1, allow_newer_file 1
[2024-08-13 12:22:09.043549] [0x00007c28] [info]    plate_to_slice=1, normative_check=1, use_first_fila_as_default=0
[2024-08-13 12:22:09.043549] [0x00007c28] [info]    read model file:C:/Users/J/Nextcloud/Imprimico/Impression/00_reference_models/03_orchid/OP035_LYRENPLANTER.3mf

[2024-08-13 12:22:09.043549] [0x00007c28] [info]    model 000000B3F22CD5C0, id 3, backup_path empty, set to C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3
[2024-08-13 12:22:09.043549] [0x00007c28] [info]    create /3D/Objects in "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3"
[2024-08-13 12:22:09.044551] [0x00007c28] [info]    create /Metadata in "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3"
[2024-08-13 12:22:09.044551] [0x00007c28] [info]    create /lock.txt in "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3"
[2024-08-13 12:22:09.044551] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1516import 3mf IMPORT_STAGE_OPEN, m_load_restore=0

[2024-08-13 12:22:09.045550] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1552import 3mf IMPORT_STAGE_READ_FILES

[2024-08-13 12:22:09.225108] [0x00007c28] [trace]   design_info, load_3mf found Copyright =
[2024-08-13 12:22:09.225108] [0x00007c28] [trace]   design_info, load_3mf found description =
[2024-08-13 12:22:09.225108] [0x00007c28] [trace]   design_info, load_3mf found designer =
[2024-08-13 12:22:09.225108] [0x00007c28] [trace]   design_info, load_3mf found designer_cover =
[2024-08-13 12:22:09.225108] [0x00007c28] [trace]   design_info, load_3mf found designer_user_id = 3441960122
[2024-08-13 12:22:09.225108] [0x00007c28] [trace]   design_info, load_3mf found license =
[2024-08-13 12:22:09.225108] [0x00007c28] [trace]   design_info, load_3mf found model name =
[2024-08-13 12:22:09.225108] [0x00007c28] [trace]   design_info, load_3mf found model name =
[2024-08-13 12:22:09.225108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 1th file [Content_Types].xml, total=19
[2024-08-13 12:22:09.225108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1795, [Content_Types].xml skipped, already parsed or a directory or not supported

[2024-08-13 12:22:09.225108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 2th file Metadata/plate_1.png, total=19
[2024-08-13 12:22:09.225108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1795, Metadata/plate_1.png skipped, already parsed or a directory or not supported

[2024-08-13 12:22:09.225108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 3th file Metadata/plate_1_small.png, total=19
[2024-08-13 12:22:09.225108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1795, Metadata/plate_1_small.png skipped, already parsed or a directory or not supported
[... 45 lines cut ...]
[2024-08-13 12:22:09.226109] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1795, _rels/.rels skipped, already parsed or a directory or not supported

[2024-08-13 12:22:09.226109] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1847, process group colors, size 0

[2024-08-13 12:22:09.226109] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1862, begin to assemble objects, size 2

[2024-08-13 12:22:09.226109] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1880, could not find object 2 in plate 1, skip it

[2024-08-13 12:22:09.226109] [0x00007c28] [debug]   Slic3r::_BBS_3MF_Importer::_generate_volumes_new: line 4393, subobject_id 3, shared_mesh_id 3
[2024-08-13 12:22:09.371107] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:2055import 3mf IMPORT_STAGE_LOADING_PLATES, m_plater_data size 2, m_backup_path C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3

[2024-08-13 12:22:09.371107] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:2100, plate 1, thumbnail_file=C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3/Metadata/plate_1.png, no_light_thumbnail_file=C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3/Metadata/plate_no_light_1.png
[2024-08-13 12:22:09.371107] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:2101, top_thumbnail_file=C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3/Metadata/top_1.png, pick_thumbnail_file=C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3/Metadata/pick_1.png
[2024-08-13 12:22:09.372108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:2100, plate 2, thumbnail_file=C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3/Metadata/plate_2.png, no_light_thumbnail_file=C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3/Metadata/plate_no_light_2.png
[2024-08-13 12:22:09.372108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:2101, top_thumbnail_file=C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3/Metadata/top_2.png, pick_thumbnail_file=C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Tue_Aug_13/12_22_09#51656#3/Metadata/pick_2.png
[2024-08-13 12:22:09.372108] [0x00007c28] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:2158import 3mf IMPORT_STAGE_FINISH

[2024-08-13 12:22:09.375107] [0x00007c28] [info]    the first file is a 3mf, version 2.1.0.1, got plate count 2
Segmentation fault
```

### Checklist of files to include

- [X] Log file
- [ ] Project file

ttt888zzz, 2024-08-16:
HI[uSyzex](https://github.com/uSyzex),Could you please upload the 3mf file and stl file

uSyzex, 2024-08-18:
> HI,[uSyzex](https://github.com/uSyzex) Could you please provide the 3mf file and stl file，thanks

Any stl / 3mf would work and I'm not allowed to share the stl I worked with.

lanewei120, 2024-08-19:
> cuments\MobaXterm

we are using cli for our internal test, and can not meet your issue here
[image]
1. could you confirm whether all the 3mfs have this issue?
if not, please upload your 3mf to us, and if it is not allowed for public, could you share it by email?
2. is your binary built by yourself or not

uSyzex, 2024-08-19:
> we are using cli for our internal test, and can not meet your issue here [...]

1. All 3mfs I have tried gave me a segfault, even when created a 3mf from bambu by adding a cube primitive and saving it to 3mf. 
2. No, I have downloaded the .exe on the Bambulab Official website. 

I'm on Windows by the way, not linux. 

Logs from the newly created 3mf:
```
⮞ ./bambu-studio.exe  --slice 01 --allow-newer-file --export-3mf Cube.3mf --debug 5
[2024-08-19 10:23:36.261920] [0x00004178] [trace]   Initializing StaticPrintConfigs
[2024-08-19 10:23:36.275920] [0x00004178] [warning] cli mode, Current BambuStudio Version 01.09.03.50
[2024-08-19 10:23:36.275920] [0x00004178] [info]    Will start to read model file now, file count :0

[2024-08-19 10:23:36.275920] [0x00004178] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0, skip_useless_pick 1, allow_newer_file 1
[2024-08-19 10:23:36.275920] [0x00004178] [info]    plate_to_slice=1, normative_check=1, use_first_fila_as_default=0
[2024-08-19 10:23:36.275920] [0x00004178] [warning] 1652: not support to slice plate 1, reset to 0
[2024-08-19 10:23:36.275920] [0x00004178] [info]    Slic3r::CLI::run:before load settings, file count=0

[2024-08-19 10:23:36.275920] [0x00004178] [info]    current printer , new printer , current process , new process
[2024-08-19 10:23:36.275920] [0x00004178] [info]    current printer inherits , new printer inherits , current process inherits , new process inherits
[2024-08-19 10:23:36.275920] [0x00004178] [info]    old 3mf, no compatible printers, set to compatible
[2024-08-19 10:23:36.275920] [0x00004178] [info]    old printer , inherited from , old process , inherited from  ,compatible 1
[2024-08-19 10:23:36.275920] [0x00004178] [warning] no filament colors found in projects
[2024-08-19 10:23:36.275920] [0x00004178] [info]    total 0 models, 0 objects

[2024-08-19 10:23:36.276923] [0x00004178] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from -2026419200 to 0
[2024-08-19 10:23:36.276923] [0x00004178] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from 0 to 1
[2024-08-19 10:23:36.276923] [0x00004178] [info]    will start transforms, commands count 0

[2024-08-19 10:23:36.276923] [0x00004178] [info]    finished model pre-process commands

[2024-08-19 10:23:36.276923] [0x00004178] [info]    before arrange, need_arrange=1, duplicate_count 0, filament_color_changed 0
[2024-08-19 10:23:36.276923] [0x00004178] [info]    Need to slice for plate 1, total plate count 1 partplates!

[2024-08-19 10:23:36.277922] [0x00004178] [info]    Slic3r::GUI::PartPlateList::store_to_3mf_structure: plate 1 before load, width 0, height 0, size 0!
[2024-08-19 10:23:36.277922] [0x00004178] [info]    Slic3r::GUI::PartPlateList::store_to_3mf_structure: plate 1 after load, width 0, height 0, size 0!
[2024-08-19 10:23:36.277922] [0x00004178] [info]    Slic3r::GUI::PartPlateList::store_to_3mf_structure: plate 0, gcode_filename=, with_slice_info=1, slice_valid 0, object item count 0.
[2024-08-19 10:23:36.277922] [0x00004178] [info]    Slic3r::GUI::PartPlateList::store_to_3mf_structure:stored 1 plates!
[2024-08-19 10:23:36.277922] [0x00004178] [info]    thumbnails stage: plate 1's thumbnail data is invalid, check the file  exist or not
[2024-08-19 10:23:36.277922] [0x00004178] [info]    thumbnails stage: plate 1's thumbnail file also not there, need to regenerate
[2024-08-19 10:23:36.277922] [0x00004178] [info]    thumbnails stage: plate 1's no_light_thumbnail_file  also not there, need to regenerate
[2024-08-19 10:23:36.277922] [0x00004178] [info]    thumbnails stage: plate 1's top_file  also not there, need to regenerate
[2024-08-19 10:23:36.277922] [0x00004178] [info]    opengl version 3.3.7
[2024-08-19 10:23:36.507463] [0x00004178] [info]    glfwInit Success.

[2024-08-19 10:23:36.664255] [0x00004178] [info]    Found Framebuffer Type ARB.

[2024-08-19 10:23:36.664255] [0x00004178] [info]    got opengl version 4.6.0 NVIDIA 552.22, glsl version 4.60 NVIDIA, vendor NVIDIA Corporation

[2024-08-19 10:23:36.670256] [0x00004178] [info]    Slic3r::GUI::OpenGLManager::init_glnot AMD driver.
[2024-08-19 10:23:36.670256] [0x00004178] [info]    glewInit Sucess.

Segmentation fault
```

lanewei120, 2024-08-19:
> 1. All 3mfs I have tried gave me a segfault, even when created a 3mf from bambu by adding a cube primitive and saving it to 3mf.
> 2. No, I have downloaded the .exe on the Bambulab Official website. [...]

 ./bambu-studio.exe  --slice 01 --allow-newer-file --export-3mf Cube.3mf --debug 5

the above command misses the input file,

uSyzex, 2024-08-19:
> > > > cuments\MobaXterm
> > > 
> > > 
> > > we are using cli for our internal test, and can not meet your issue here ![image](https://private-user-images.githubusercontent.com/106916061/358950549-1603eb84-f1b5-48b4-8001-8a422f00092a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQwNTU5MDIsIm5iZiI6MTcyNDA1NTYwMiwicGF0aCI6Ii8xMDY5MTYwNjEvMzU4OTUwNTQ5LTE2MDNlYjg0LWYxYjUtNDhiNC04MDAxLThhNDIyZjAwMDkyYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxOVQwODIwMDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lZjIwM2VmMGJiN2E2MzU3MjA0MzhmZWNkZGJkOTQ1MjZiZWM0OTg4ZjJjOTEyYzUyNzUzZmQzNTQ2MzI5MTlmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kF_3NXYE6PsnOhOijobjpL62lw1t4AZRcf5YTPM0qBM)
> > > ```
> > > 1. could you confirm whether all the 3mfs have this issue?
> > >    if not, please upload your 3mf to us, and if it is not allowed for public, could you share it by email?
> > > 
> > > 2. is your binary built by yourself or not
> > > ```
> > 
> > 
> > 
> > 1. All 3mfs I have tried gave me a segfault, even when created a 3mf from bambu by adding a cube primitive and saving it to 3mf.
> > 2. No, I have downloaded the .exe on the Bambulab Official website.
> > 
> > I'm on Windows by the way, not linux.
> > Logs from the newly created 3mf:
> > ```
> > ⮞ ./bambu-studio.exe  --slice 01 --allow-newer-file --export-3mf Cube.3mf --debug 5
> > [2024-08-19 10:23:36.261920] [0x00004178] [trace]   Initializing StaticPrintConfigs
> > [2024-08-19 10:23:36.275920] [0x00004178] [warning] cli mode, Current BambuStudio Version 01.09.03.50
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    Will start to read model file now, file count :0
> > 
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0, skip_useless_pick 1, allow_newer_file 1
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    plate_to_slice=1, normative_check=1, use_first_fila_as_default=0
> > [2024-08-19 10:23:36.275920] [0x00004178] [warning] 1652: not support to slice plate 1, reset to 0
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    Slic3r::CLI::run:before load settings, file count=0
> > 
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    current printer , new printer , current process , new process
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    current printer inherits , new printer inherits , current process inherits , new process inherits
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    old 3mf, no compatible printers, set to compatible
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    old printer , inherited from , old process , inherited from  ,compatible 1
> > [2024-08-19 10:23:36.275920] [0x00004178] [warning] no filament colors found in projects
> > [2024-08-19 10:23:36.275920] [0x00004178] [info]    total 0 models, 0 objects
> > 
> > [2024-08-19 10:23:36.276923] [0x00004178] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from -2026419200 to 0
> > [2024-08-19 10:23:36.276923] [0x00004178] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from 0 to 1
> > [2024-08-19 10:23:36.276923] [0x00004178] [info]    will start transforms, commands count 0
> > 
> > [2024-08-19 10:23:36.276923] [0x00004178] [info]    finished model pre-process commands
> > 
> > [2024-08-19 10:23:36.276923] [0x00004178] [info]    before arrange, need_arrange=1, duplicate_count 0, filament_color_changed 0
> > [2024-08-19 10:23:36.276923] [0x00004178] [info]    Need to slice for plate 1, total plate count 1 partplates!
> > 
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    Slic3r::GUI::PartPlateList::store_to_3mf_structure: plate 1 before load, width 0, height 0, size 0!
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    Slic3r::GUI::PartPlateList::store_to_3mf_structure: plate 1 after load, width 0, height 0, size 0!
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    Slic3r::GUI::PartPlateList::store_to_3mf_structure: plate 0, gcode_filename=, with_slice_info=1, slice_valid 0, object item count 0.
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    Slic3r::GUI::PartPlateList::store_to_3mf_structure:stored 1 plates!
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    thumbnails stage: plate 1's thumbnail data is invalid, check the file  exist or not
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    thumbnails stage: plate 1's thumbnail file also not there, need to regenerate
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    thumbnails stage: plate 1's no_light_thumbnail_file  also not there, need to regenerate
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    thumbnails stage: plate 1's top_file  also not there, need to regenerate
> > [2024-08-19 10:23:36.277922] [0x00004178] [info]    opengl version 3.3.7
> > [2024-08-19 10:23:36.507463] [0x00004178] [info]    glfwInit Success.
> > 
> > [2024-08-19 10:23:36.664255] [0x00004178] [info]    Found Framebuffer Type ARB.
> > 
> > [2024-08-19 10:23:36.664255] [0x00004178] [info]    got opengl version 4.6.0 NVIDIA 552.22, glsl version 4.60 NVIDIA, vendor NVIDIA Corporation
> > 
[... cut ...] :GUI::OpenGLManager::init_glnot AMD driver.
> > [2024-08-19 10:23:36.670256] [0x00004178] [info]    glewInit Sucess.
> > 
> > Segmentation fault
> > ```
> 
> ./bambu-studio.exe --slice 01 --allow-newer-file --export-3mf Cube.3mf --debug 5
> 
> the above command misses the input file, "Cube.3mf" is used as the output file not input file

Oups that's my bad.
It actually works this time. 

But when using my personal 3mf, it gives me a segfault. Can you give me the email I need to send you the file.

lanewei120, 2024-08-19:
lane.wei@bambulab.com
I will delete this address as soon as I received your email

uSyzex, 2024-08-19:
> your 3mf has been modified by somewhere, not generated by bambustudio directly
> please use bambu-studio to regenerate the 3mf and pass it to bambu-studio.exe for cli slicing
> it can not be processed by bambu-studio.exe currently

It has been generated with Orca Slicer in fact. Even if bambu studio is not able to support it, it should throw a different error than a segfault.

-------- Original Message --------