bambulab/BambuStudio #4628: Scaling option doesn't work in BambuStudio CLI

uSyzex, 2024-08-13:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

i7 6700k, 32GB RAM, 3070Ti

### Printer

X1C

### How to reproduce

Since 3mf files lead to a Segfault #4627, I will be using STLs to show the issue. 

1. `./bambu-studio.exe --scale 2 --info --allow-newer-file model_test.stl --debug 5`
OR 
2. ./bambu-studio.exe --scale 2 --slice 1 --allow-newer-file model_test.stl --debug 5

### Actual results

Doesn't do anything

### Expected results

Scale the STL & process the action, info or slice. 

### Project file & Debug log uploads

```
[2024-08-13 12:53:48.911460] [0x000032a4] [trace]   Initializing StaticPrintConfigs
[2024-08-13 12:53:48.925457] [0x000032a4] [warning] cli mode, Current BambuStudio Version 01.09.03.50
[2024-08-13 12:53:48.925457] [0x000032a4] [info]    Will start to read model file now, file count :1

[2024-08-13 12:53:48.926456] [0x000032a4] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0, skip_useless_pick 1, allow_newer_file 1
[2024-08-13 12:53:48.926456] [0x000032a4] [info]    plate_to_slice=1, normative_check=1, use_first_fila_as_default=0
[2024-08-13 12:53:48.926456] [0x000032a4] [info]    read model file:test_model.stl

[2024-08-13 12:53:48.982455] [0x000032a4] [debug]   TriangleMesh::repair() started
[2024-08-13 12:53:48.985457] [0x000032a4] [debug]   TriangleMesh::repair() finished
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    object test_model.stl, id :5, from stl or other 3mf

[2024-08-13 12:53:48.991456] [0x000032a4] [warning] 1652: not support to slice plate 1, reset to 0
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    Slic3r::CLI::run:before load settings, file count=0

[2024-08-13 12:53:48.991456] [0x000032a4] [info]    current printer , new printer , current process , new process
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    current printer inherits , new printer inherits , current process inherits , new process inherits
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    old 3mf, no compatible printers, set to compatible
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    old printer , inherited from , old process , inherited from  ,compatible 1
[2024-08-13 12:53:48.991456] [0x000032a4] [warning] no filament colors found in projects
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    total 1 models, 0 objects

[2024-08-13 12:53:48.991456] [0x000032a4] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from 140 to 0
[2024-08-13 12:53:48.991456] [0x000032a4] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from 0 to 1
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    translate_old 0, shrink_to_new_bed 0, old bed size {200, 200, 100}
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    Slic3r::GUI::PartPlateList::reset_size:before size: plate_width 0, plate_depth 0, plate_height 0
[2024-08-13 12:53:48.991456] [0x000032a4] [info]    Slic3r::GUI::PartPlateList::reset_size:after size: plate_width 200, plate_depth 200, plate_height 100
[2024-08-13 12:53:48.992455] [0x000032a4] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: plate_id 0, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2024-08-13 12:53:48.992455] [0x000032a4] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: with_instance_move 0, after, origin {0,0,0}, plate_width 200, plate_depth 200, plate_height 100
[2024-08-13 12:53:48.992455] [0x000032a4] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-13 12:53:48.992455] [0x000032a4] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: plate_id 1, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2024-08-13 12:53:48.992455] [0x000032a4] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: with_instance_move 0, after, origin {240,0,0}, plate_width 200, plate_depth 200, plate_height 100
[2024-08-13 12:53:48.992455] [0x000032a4] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-13 12:53:48.992455] [0x000032a4] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-13 12:53:48.992455] [0x000032a4] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-13 12:53:48.992455] [0x000032a4] [error]   Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles

[2024-08-13 12:53:48.992455] [0x000032a4] [info]    will start transforms, commands count 1

[2024-08-13 12:53:48.992455] [0x000032a4] [info]    process transform scale

[2024-08-13 12:53:50.302240] [0x000032a4] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-13 12:53:50.302240] [0x000032a4] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-13 12:53:50.302240] [0x000032a4] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-13 12:53:50.302240] [0x000032a4] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0

```

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-08-15:
Hello, can you record a GIF? Thank you

uSyzex, 2024-08-18:
> Hello, can you record a GIF? Thank you

A GIF would not had any more information to the debug logs I added.

lanewei120, 2024-08-19:
hello,

how do you judge the scaling has no effect?

could you use --export-3mf option to output the result to a 3mf and check whether scaled there?
[image]

uSyzex, 2024-08-19:
> hello,
>
> how do you judge the scaling has no effect?
>
> could you use --export-3mf option to output the result to a 3mf and check whether scaled there? [...]

Because the given action is not executed, --info or --slice. 

When using --export-3mf the file is not created and the logs still mention an error
`[2024-08-19 10:15:10.156560] [0x00004ed0] [error]   Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles`

Full logs:
```
[J.DESKTOP-OKA88J7] ⮞ ./bambu-studio.exe --scale 2 --slice 1 --allow-newer-file --export-3mf test.3mf test_model.stl --debug 5
[2024-08-19 10:15:10.079560] [0x00004ed0] [trace]   Initializing StaticPrintConfigs
[2024-08-19 10:15:10.094560] [0x00004ed0] [warning] cli mode, Current BambuStudio Version 01.09.03.50
[2024-08-19 10:15:10.094560] [0x00004ed0] [info]    Will start to read model file now, file count :1

[2024-08-19 10:15:10.094560] [0x00004ed0] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0, skip_useless_pick 1, allow_newer_file 1
[2024-08-19 10:15:10.094560] [0x00004ed0] [info]    plate_to_slice=1, normative_check=1, use_first_fila_as_default=0
[2024-08-19 10:15:10.094560] [0x00004ed0] [info]    read model file:test_model.stl

[2024-08-19 10:15:10.150559] [0x00004ed0] [debug]   TriangleMesh::repair() started
[2024-08-19 10:15:10.152560] [0x00004ed0] [debug]   TriangleMesh::repair() finished
[2024-08-19 10:15:10.155559] [0x00004ed0] [info]    object test_model.stl, id :5, from stl or other 3mf

[2024-08-19 10:15:10.156560] [0x00004ed0] [warning] 1652: not support to slice plate 1, reset to 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::CLI::run:before load settings, file count=0

[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    current printer , new printer , current process , new process
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    current printer inherits , new printer inherits , current process inherits , new process inherits
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    old 3mf, no compatible printers, set to compatible
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    old printer , inherited from , old process , inherited from  ,compatible 1
[2024-08-19 10:15:10.156560] [0x00004ed0] [warning] no filament colors found in projects
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    total 1 models, 0 objects

[2024-08-19 10:15:10.156560] [0x00004ed0] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from -1610612736 to 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from 0 to 1
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    translate_old 0, shrink_to_new_bed 0, old bed size {200, 200, 100}
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlateList::reset_size:before size: plate_width 0, plate_depth 0, plate_height 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlateList::reset_size:after size: plate_width 200, plate_depth 200, plate_height 100
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: plate_id 0, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: with_instance_move 0, after, origin {0,0,0}, plate_width 200, plate_depth 200, plate_height 100
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: plate_id 1, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: with_instance_move 0, after, origin {240,0,0}, plate_width 200, plate_depth 200, plate_height 100
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-19 10:15:10.156560] [0x00004ed0] [error]   Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles

[2024-08-19 10:15:10.157561] [0x00004ed0] [info]    will start transforms, commands count 1

[2024-08-19 10:15:10.157561] [0x00004ed0] [info]    process transform scale

[2024-08-19 10:15:11.400136] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-19 10:15:11.400136] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-19 10:15:11.400136] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-19 10:15:11.400136] [0x00004ed0] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
```

Can you provide me the test model you are using so I can test on my end with it ?

uSyzex, 2024-08-19:
When using the --export-3mf on a newly created 3mf, it stills doesn't work:
The new 3mf was created with Bambustudio by addid a cube primitive and saving it to 3mf.

```
⮞ ./bambu-studio.exe --scale 2  --allow-newer-file --export-3mf D:/Document/Cubescaled.3mf Cube.3mf --debug 5
[2024-08-19 10:28:51.801769] [0x00001aa8] [trace]   Initializing StaticPrintConfigs
[2024-08-19 10:28:51.815767] [0x00001aa8] [warning] cli mode, Current BambuStudio Version 01.09.03.50
[2024-08-19 10:28:51.815767] [0x00001aa8] [info]    Will start to read model file now, file count :1

[2024-08-19 10:28:51.815767] [0x00001aa8] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0, skip_useless_pick 1, allow_newer_file 1
[2024-08-19 10:28:51.815767] [0x00001aa8] [info]    plate_to_slice=0, normative_check=1, use_first_fila_as_default=0
[2024-08-19 10:28:51.815767] [0x00001aa8] [info]    read model file:Cube.3mf

[2024-08-19 10:28:51.815767] [0x00001aa8] [info]    model 000000B471CCDC00, id 3, backup_path empty, set to C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3
[2024-08-19 10:28:51.815767] [0x00001aa8] [info]    create /3D/Objects in "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3"
[2024-08-19 10:28:51.815767] [0x00001aa8] [info]    create /Metadata in "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3"
[2024-08-19 10:28:51.816768] [0x00001aa8] [info]    create /lock.txt in "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3"
[2024-08-19 10:28:51.816768] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1516import 3mf IMPORT_STAGE_OPEN, m_load_restore=0

[2024-08-19 10:28:51.816768] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1552import 3mf IMPORT_STAGE_READ_FILES

[2024-08-19 10:28:51.817768] [0x00001aa8] [trace]   design_info, load_3mf found Copyright =
[2024-08-19 10:28:51.817768] [0x00001aa8] [trace]   design_info, load_3mf found description =
[2024-08-19 10:28:51.817768] [0x00001aa8] [trace]   design_info, load_3mf found designer =
[2024-08-19 10:28:51.817768] [0x00001aa8] [trace]   design_info, load_3mf found designer_cover =
[2024-08-19 10:28:51.817768] [0x00001aa8] [trace]   design_info, load_3mf found designer_user_id = 3441960122
[2024-08-19 10:28:51.817768] [0x00001aa8] [trace]   design_info, load_3mf found license =
[2024-08-19 10:28:51.817768] [0x00001aa8] [trace]   design_info, load_3mf found model name =
[2024-08-19 10:28:51.817768] [0x00001aa8] [trace]   design_info, load_3mf found model name =
[2024-08-19 10:28:51.817768] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 1th file [Content_Types].xml, total=14
[2024-08-19 10:28:51.817768] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1795, [Content_Types].xml skipped, already parsed or a directory or not supported

[2024-08-19 10:28:51.817768] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 2th file Metadata/plate_1.png, total=14
[2024-08-19 10:28:51.819769] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_extract_file_from_archive, extract  "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3/Metadata/plate_1.png" from 3mf Metadata/plate_1.png, ret 1

[2024-08-19 10:28:51.819769] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 3th file Metadata/plate_1_small.png, total=14
[2024-08-19 10:28:51.825769] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_extract_file_from_archive, extract  "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3/Metadata/plate_1_small.png" from 3mf Metadata/plate_1_small.png, ret 1

[2024-08-19 10:28:51.825769] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 4th file Metadata/plate_no_light_1.png, total=14
[2024-08-19 10:28:51.827772] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_extract_file_from_archive, extract  "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3/Metadata/plate_no_light_1.png" from 3mf Metadata/plate_no_light_1.png, ret 1

[2024-08-19 10:28:51.827772] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 5th file Metadata/top_1.png, total=14
[2024-08-19 10:28:51.829772] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_extract_file_from_archive, extract  "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3/Metadata/top_1.png" from 3mf Metadata/top_1.png, ret 1

[2024-08-19 10:28:51.829772] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 6th file Metadata/pick_1.png, total=14
[2024-08-19 10:28:51.830772] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_extract_file_from_archive, extract  "C:\Users\J\Documents\MobaXterm\slash\tmp/bamboo_model/Mon_Aug_19/10_28_51#3996#3/Metadata/pick_1.png" from 3mf Metadata/pick_1.png, ret 1

[2024-08-19 10:28:51.830772] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1705extract 7th file 3D/3dmodel.model, total=14
[2024-08-19 10:28:51.830772] [0x00001aa8] [info]    Slic3r::_BBS_3MF_Importer::_load_model_from_file:1795, 3D/3dmodel.model skipped, already parsed or a directory or not supported

[... cut ...]
[2024-08-19 10:28:53.112142] [0x00001aa8] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-19 10:28:53.112142] [0x00001aa8] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-19 10:28:53.113145] [0x00001aa8] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-19 10:28:53.113145] [0x00001aa8] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
```

lanewei120, 2024-08-19:
just found a bug when doing scale under cli mode
we will fix it

uSyzex, 2024-08-23:
> just found a bug when doing scale under cli mode we will fix it

Do you have any ETA ? Thanks

lanewei120, 2024-08-24:
we will submit this fix soon, maybe next week

thanks

lanewei120, 2024-08-30:
already included in 1.9.5

closed 2024-08-30 by lanewei120