bambulab/BambuStudio #4632: Moving part along Z axis gives unexpected and incorrect results

mark6B, 2024-08-14:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home Version 23H2 OS Build 22631.4037

### Additional system information

Processor	13th Gen Intel(R) Core(TM) i7-1355U   1.70 GHz
Installed RAM	16.0 GB (15.7 GB usable)
Device ID	3490A255-1DD5-485F-99AA-0C0F72AC4BB7
Product ID	00342-22281-37757-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 10 touch points

### Printer

Bambu Lab P1S Combo (w/ AMS)

### How to reproduce

1. Add a model to the print plate
2. Add a part to the model
3. Attempt to move the part along the Z axis using world coordinates to 3.31mm (0.01mm above top surface)

### Actual results

Part moves to 3.29mm along the Z axis.  Attempting to move it to 3.31mm along the Z axis again results in the part moving to 3.27mm Z.

### Expected results

Part should move to 3.31mm along the Z axis.

### Project file & Debug log uploads

[Z Axis Move Issue.zip](https://github.com/user-attachments/files/16616874/Z.Axis.Move.Issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-08-15:
Hello, can you record a GIF? Thank you

mark6B, 2024-08-15:
I did a screen recording and have attached it here.
[Screen Recording 2024-08-14 132854.zip](https://github.com/user-attachments/files/16626411/Screen.Recording.2024-08-14.132854.zip)

Haidiye00, 2024-08-19:
[image]
It should be that the coordinate system is incorrect. We have recently added a translation tool for local coordinate systems, which is currently undergoing internal testing. Please look forward to it~

mark6B, 2024-08-19:
I'm looking forward to the new feature, however, it seems there is a bug
with this capability.  I can move text along the X and Y axis to exactly
where I want it to be.  However, moving parts along the Z axis does not
work correctly.  Regardless of the coordinate system, I should be able to
specify an exact X,Y,Z location for a part.  Today, I have to use the mouse
and struggle to get it "close enough" so it looks good.  If I can enter
X,Y,Z coordinates, I can put the part exactly where I want it every time
quickly and easily.

That's the key point.  If I enter 3.31mm along the Z axis, the part should
move to 3.31mm, not 3.29mm.  If I try again to move to 3.31mm, it moves to
3.27mm.  So, something is wrong.

drekahedron, 2024-09-17:
Commenting here that I've also encountered this issue, running Bambu 1.9.5.51. Trying to move a part along the Z axis, just simply to 0.0mm, and it keeps reducing the current position by an inconsistent amount, approximately 0.02mm at a time.

Haidiye00, 2024-09-18:
Sorry, the new feature is expected to appear in version 1.10 and will not be available in version 1.9 at the moment.

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00

mark6B, 2024-10-17:
Where can I get the beta version?  I just checked the site, but cannot find
it.

Haidiye00, 2024-10-18:
https://github.com/bambulab/BambuStudio/releases