bambulab/BambuStudio #4633: Thumbnail dont work in Bambu Studio

Zoeconner1, 2024-08-15:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macos

### Printer

A1

### How to reproduce

use the app

### Actual results

You dont see the thumbnails on the page

### Expected results

look at it

### Project file & Debug log uploads

ergv

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-08-16:
HI，[Zoeconner1](https://github.com/Zoeconner1)

Zoeconner1, 2024-08-16:
[image]

Zoeconner1, 2024-08-17:
[image]

qian0807, 2024-08-19:
HI, [Zoeconner1](https://github.com/Zoeconner1), Please confirm your Safari browser version. If the Safari browser version is 13 or below, it will cause webp format images to fail to load and the content layout to display incorrectly.

Zoeconner1, 2024-08-19:
Using Google as main browser

qian0807, 2024-08-21:
> Using Google as main browser

Hello, the browser used here is only related to the current computer's operating system and has nothing to do with the default browser settings. On macOS, Safari is used.

Bumblebeats, 2025-01-12:
I'm having this same, issue.  What was the resolution?
Running on a mid-2012 Macbook pro, OS 10.15.7, Safari 15.6.1
Loading MakerWorld in the Safari webbrowser also has the same issues of not showing the WebP images

Zoeconner1, 2025-01-12:
Pc to old