bambulab/BambuStudio #4634: Segmentation fault (core dumped) immediately upon startup

PollastreGH, 2024-08-15:
### Bambu Studio Version

1.9.3

### Where is the application from?

Bambu Lab github releases, Built from source by yourself or others

### OS version

EndeavourOS (Linux 6.10.4.arch2-1)

### Additional system information

Desktop Environment: KDE Plasma
Compositor: Wayland (the bug remains on X11)

### Printer

Bambu P1S

### How to reproduce

1. Attempt to launch Bambu Studio via either the desktop interface or by running bambu-studio in the terminal emulator.
2. The program will output:
```
[2024-08-14 23:48:12.146858] [0x0000712dab64f480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
Segmentation fault (core dumped)
```
3. The program crashes

### Actual results

The program experiences an error stating `Segmentation fault (core dumped)` and crashes

### Expected results

The program is expected to launch succesfully

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16621871/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

IvanVojtko, 2024-08-15:
Same problem for me using Arch Linux.

dmophir, 2024-08-15:
I'm also having the same issue on Arch Linux with the same console output:

```
[2024-08-15 10:58:38.088873] [0x0000766fb6100480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
Segmentation fault (core dumped)
```
### Bambu Studio Version
1.09.03.50-1

### Where is the application from?
I experienced the same behavior using Bambu Studio from the following sources:
* [Arch User Repository](https://aur.archlinux.org/packages/bambustudio-bin)
* [Bambu_Studio_linux_ubuntu-v01.09.03.50.AppImage ](https://github.com/bambulab/BambuStudio/releases/download/v01.09.03.50/Bambu_Studio_linux_ubuntu-v01.09.03.50.AppImage) 
* [Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage](https://github.com/bambulab/BambuStudio/releases/download/v01.09.03.50/Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage)

From each application source, including the official BambuStudio GitHub release page, I experienced the same issue.

### OS Version
Arch Linux, kernel 6.10.4-arch2-1
GNOME 46.4 under X11

### Additional System Information
Ryzen 5900X
ASRock x570 ITX/TB
Nvidia RTX 3090 Founders Edition
64GB RAM

### Printer
Bambu X1 Carbon

### How To Reproduce
1. Launch Bambu Studio via command line
2. Output will be as above

### Actual results
Program crashes after above output

### Expected results
Program launches successfully

### Project file & Debug Log uploads
Core dump via Systemd-Journal
```
Aug 15 11:04:52 da6 kernel: bambustu_main[59301]: segfault at 22 ip 000061d93d5da874 sp 00007ffd42881aa8 error 4 in bambu-studio[37a0874,61d93a3d4000+3db1000] likely on CPU 6 >
Aug 15 11:04:52 da6 kernel: Code: 0f 83 48 ff ff ff 44 89 d3 eb bf 0f 1f 00 48 03 45 10 e9 d3 fe ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 f3 0f 1e fa <44> 8b 46 04 4>
Aug 15 11:04:52 da6 gnome-shell[1607]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x5000006
Aug 15 11:04:52 da6 systemd-coredump[59317]: Process 59301 (bambustu_main) of user 1000 terminated abnormally with signal 11/SEGV, processing...
Aug 15 11:04:52 da6 systemd[1]: Started Process Core Dump (PID 59317/UID 0).
Aug 15 11:04:53 da6 systemd-coredump[59318]: [🡕] Process 59301 (bambustu_main) of user 1000 dumped core.
                                             
                                             Stack trace of thread 59301:
                                             #0  0x000061d93d5da874 tt_face_get_paint_layers (/tmp/.mount_Bambu_GMaPnc/bin/bambu-studio + 0x37a0874)
                                             #1  0x00007a28c0b6778d n/a (libcairo.so.2 + 0x9a78d)
                                             #2  0x00007a28c0b1dd00 n/a (libcairo.so.2 + 0x50d00)
                                             #3  0x00007a28c0b1df87 cairo_scaled_font_glyph_extents (libcairo.so.2 + 0x50f87)
                                             #4  0x00007a28c18e0242 n/a (libpangocairo-1.0.so.0 + 0x8242)
                                             #5  0x00007a28c13d1779 pango_glyph_string_extents_range (libpango-1.0.so.0 + 0x15779)
                                             #6  0x00007a28c13ea01a n/a (libpango-1.0.so.0 + 0x2e01a)
                                             #7  0x00007a28c13eb253 n/a (libpango-1.0.so.0 + 0x2f253)
                                             #8  0x00007a28c13dfb46 n/a (libpango-1.0.so.0 + 0x23b46)
                                             #9  0x00007a28c13dff6c n/a (libpango-1.0.so.0 + 0x23f6c)
                                             #10 0x00007a28c13e026e pango_layout_get_size (libpango-1.0.so.0 + 0x2426e)
                                             #11 0x00007a28c13ee827 n/a (libpango-1.0.so.0 + 0x32827)
                                             #12 0x00007a28c13e009f n/a (libpango-1.0.so.0 + 0x2409f)
                                             #13 0x00007a28c13e41b4 pango_layout_get_pixel_size (libpango-1.0.so.0 + 0x281b4)
                                             #14 0x000061d93ddae058 _ZNK14wxCairoContext13GetTextExtentERK8wxStringPdS3_S3_S3_ (/tmp/.mount_Bambu_GMaPnc/bin/bambu-studio + 0x3>
                                             #15 0x000061d93db50167 _ZNK10wxGCDCImpl15DoGetTextExtentERK8wxStringPiS3_S3_S3_PK6wxFont (/tmp/.mount_Bambu_GMaPnc/bin/bambu-studi>
                                             #16 0x000061d93b241eac _ZN6Slic3r3GUI15BBLSplashScreen8DecorateER8wxBitmap (/tmp/.mount_Bambu_GMaPnc/bin/bambu-studio + 0x1407eac)
[... 40 lines cut ...]
                                             #1  0x00007a28c080b367 g_cond_wait_until (libglib-2.0.so.0 + 0xb8367)
                                             #2  0x00007a28c07788d5 n/a (libglib-2.0.so.0 + 0x258d5)
                                             #3  0x00007a28c07e527b n/a (libglib-2.0.so.0 + 0x9227b)
                                             #4  0x00007a28c07df266 n/a (libglib-2.0.so.0 + 0x8c266)
                                             #5  0x00007a28b92a339d n/a (libc.so.6 + 0x9439d)
                                             #6  0x00007a28b932849c n/a (libc.so.6 + 0x11949c)
                                             
                                             Stack trace of thread 59314:
                                             #0  0x00007a28b931a63d __poll (libc.so.6 + 0x10b63d)
                                             #1  0x00007a28c081192d n/a (libglib-2.0.so.0 + 0xbe92d)
[... cut ...]
                                             #6  0x00007a28b932849c n/a (libc.so.6 + 0x11949c)
                                             ELF object binary architecture: AMD x86-64
Aug 15 11:04:53 da6 systemd[1]: tmp-.mount_Bambu_GMaPnc.mount: Deactivated successfully.
Aug 15 11:04:53 da6 systemd[1]: systemd-coredump@10-59317-0.service: Deactivated successfully.
Aug 15 11:04:53 da6 systemd[1]: systemd-coredump@10-59317-0.service: Consumed 209ms CPU time, 69.1M memory peak.
```

### Checklist of files to include

- [x] Log file
- [ ] Project file

johndickinson, 2024-08-15:
I also see this with Arch Linux. The backtrace from gdb shows
```
(gdb) bt
#0  0x0000555558cf4874 in tt_face_get_paint_layers ()
#1  0x00007ffff707b78d in _cairo_ft_scaled_glyph_is_colr_v1 (scaled_font=<optimized out>, scaled_glyph=0x55555b2857d0, face=0x55555b27fdd0)
    at ../cairo/src/cairo-ft-font.c:3266
#2  _cairo_ft_scaled_glyph_init_metrics
    (foreground_color=0x7ffff70e4c00 <cairo_color_black.lto_priv>, load_flags=66048, vertical_layout=0, face=0x55555b27fdd0, scaled_glyph=0x55555b2857d0, scaled_font=<optimized out>) at ../cairo/src/cairo-ft-font.c:3335
#3  _cairo_ft_scaled_glyph_init
    (abstract_font=<optimized out>, scaled_glyph=0x55555b2857d0, info=CAIRO_SCALED_GLYPH_INFO_METRICS, foreground_color=0x7ffff70e4c00 <cairo_color_black.lto_priv>) at ../cairo/src/cairo-ft-font.c:3453
#4  0x00007ffff7031d00 in _cairo_scaled_glyph_lookup
    (scaled_font=scaled_font@entry=0x55555af294a0, index=<optimized out>, info=info@entry=CAIRO_SCALED_GLYPH_INFO_METRICS, foreground_color=0x7ffff70e4c00 <cairo_color_black.lto_priv>, foreground_color@entry=0x0, scaled_glyph_ret=scaled_glyph_ret@entry=0x7fffffff6ee0)
    at ../cairo/src/cairo-scaled-font.c:2913
#5  0x00007ffff7031f87 in cairo_scaled_font_glyph_extents
    (scaled_font=0x55555af294a0, glyphs=glyphs@entry=0x7fffffff6f30, num_glyphs=num_glyphs@entry=1, extents=extents@entry=0x7fffffff6f50)
    at ../cairo/src/cairo-scaled-font.c:1643
#6  0x00007ffff7ee9242 in compute_glyph_extents (cf_priv=0x55555b281e68, glyph=30, entry=0x55555b284260) at ../pango/pango/pangocairo-font.c:807
#7  _pango_cairo_font_private_get_glyph_extents_cache_entry (cf_priv=0x55555b281e68, glyph=30) at ../pango/pango/pangocairo-font.c:833
#8  _pango_cairo_font_private_get_glyph_extents (cf_priv=0x55555b281e68, glyph=30, ink_rect=0x7fffffff7000, logical_rect=0x7fffffff7010)
    at ../pango/pango/pangocairo-font.c:870
#9  pango_cairo_fc_font_get_glyph_extents (font=0x55555b281df0, glyph=30, ink_rect=0x7fffffff7000, logical_rect=0x7fffffff7010)
    at ../pango/pango/pangocairo-fcfont.c:130
#10 0x00007ffff7a37779 in pango_glyph_string_extents_range
    (glyphs=0x55555b280ab0, start=0, end=12, font=0x55555b281df0, ink_rect=<optimized out>, logical_rect=0x7fffffff70c0)
    at ../pango/pango/glyphstring.c:210
#11 pango_glyph_string_extents_range
    (glyphs=0x55555b280ab0, start=0, end=12, font=0x55555b281df0, ink_rect=<optimized out>, logical_rect=0x7fffffff70c0)
    at ../pango/pango/glyphstring.c:164
#12 0x00007ffff7a5001a in pango_layout_run_get_extents_and_height
    (run=0x55555b27ee70, run_ink=run_ink@entry=0x7fffffff71a0, run_logical=0x7fffffff70c0, 
    run_logical@entry=0x0, line_logical=line_logical@entry=0x7fffffff7190, height=height@entry=0x7fffffff718c)
    at ../pango/pango/pango-layout.c:5640
#13 0x00007ffff7a51253 in pango_layout_line_get_extents_and_height.part.0.lto_priv.0
    (line=0x55555b181710, ink_rect=<optimized out>, logical_rect=<optimized out>, height=<optimized out>) at ../pango/pango/pango-layout.c:5836
#14 0x00007ffff7a45b46 in pango_layout_line_get_extents_and_height
    (line=0x55555b181710, ink_rect=0x0, logical_rect=0x7fffffff7220, height=0x7fffffff7218) at ../pango/pango/pango-layout.c:5779
#15 get_line_extents_layout_coords
    (layout=layout@entry=0x55555b277110, line=0x55555b181710, layout_width=layout_width@entry=-1, y_offset=0, baseline=baseline@entry=0x7fffffff72dc, line_ink_layout=line_ink_layout@entry=0x0, line_logical_layout=0x7fffffff72e0) at ../pango/pango/pango-layout.c:2877
#16 0x00007ffff7a45f6c in pango_layout_get_extents_internal (layout=0x55555b277110, ink_rect=<optimized out>, 
    ink_rect@entry=0x0, logical_rect=<optimized out>, logical_rect@entry=0x7fffffff7350, line_extents=line_extents@entry=0x0)
    at ../pango/pango/pango-layout.c:3001
#17 0x00007ffff7a46220 in pango_layout_get_extents
    (layout=<optimized out>, ink_rect=ink_rect@entry=0x0, logical_rect=logical_rect@entry=0x7fffffff7350) at ../pango/pango/pango-layout.c:3118
#18 0x00007ffff7a4626e in pango_layout_get_size (layout=<optimized out>, width=0x7fffffff7640, height=0x7fffffff7560)
    at ../pango/pango/pango-layout.c:3166
#19 0x00007ffff7a54827 in pango_layout_check_lines.part.0.lto_priv.0 (layout=<optimized out>) at ../pango/pango/pango-layout.c:4983
#20 0x00007ffff7a4609f in pango_layout_check_lines (layout=<optimized out>) at ../pango/pango/pango-layout.c:4792
#21 pango_layout_get_extents_internal
    (layout=0x55555b277110, ink_rect=ink_rect@entry=0x0, logical_rect=logical_rect@entry=0x7fffffff7810, line_extents=line_extents@entry=0x0)
    at ../pango/pango/pango-layout.c:2925
#22 0x00007ffff7a4a1b4 in pango_layout_get_pixel_size (layout=<optimized out>, width=0x7fffffff7884, height=0x7fffffff7888)
    at ../pango/pango/pango-layout.c:3194
#23 0x00005555594c8058 in wxCairoContext::GetTextExtent(wxString const&, double*, double*, double*, double*) const ()
#24 0x000055555926a167 in wxGCDCImpl::DoGetTextExtent(wxString const&, int*, int*, int*, int*, wxFont const*) const ()
#25 0x000055555695beac in Slic3r::GUI::BBLSplashScreen::Decorate(wxBitmap&) ()
#26 0x000055555694f243 in Slic3r::GUI::GUI_App::on_init_inner() ()
#27 0x0000555556950a9a in Slic3r::GUI::GUI_App::OnInit() ()
#28 0x0000555559577ca7 in wxEntry(int&, wchar_t**) ()
#29 0x0000555556911abb in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#30 0x0000555555f6917b in Slic3r::CLI::run(int, char**) ()
#31 0x0000555555e80ba9 in main ()
```

[... cut ...]
### Bambu Studio Version

1.09.03.50-1
### Where is the application from?

[... 3 lines cut ...]

### OS Version

Arch Linux, kernel 6.10.4-arch2-1
Sway, wayland
### Additional System Information

Intel Xeon W5-3425
ASUSTeK  Pro WS W790E-SAGE SE
NVIDIA GeForce RTX 4070
64GB RAM
### Printer

Bambu P1S
### How To Reproduce

    Launch Bambu Studio via command line
    Output will be as above

### Actual results

Program crashes after above output
### Expected results

Program launches successfully

Sanaki, 2024-08-15:
Having the same issue as of today's updates on Garuda Linux (arch derivative). #3109 appears to be related, and pins the blame on the webkit2gtk update.

hyperpuncher, 2024-08-16:
I have the same error with OrcaSlicer today, downgrading **freetype2** to `2.13.2-2` fixed it for me.

johndickinson, 2024-08-16:
Downgrading to freetype 2.13.2-2 work for me too.

dmophir, 2024-08-16:
Downgrading `freetype2` to `2.13.2-2` also worked for me. Bambu Studio is now launching correctly.

Sanaki, 2024-08-16:
I can confirm the same, but of note, this is a workaround, not a solution. The core issue still needs fixing.

bttger, 2024-08-23:
Same issue for me.

Sanaki, 2024-08-28:
Looks like the [AUR bambustudio-bin](https://aur.archlinux.org/packages/bambustudio-bin) was updated to v01.09.04.60, which allows the program to start up again. Can't find the commit responsible for fixing things, but assuming this is an unmodified appimage, the issue should be fixed in nightly builds.

MackBambu, 2024-09-06:
Does this problem still exist on 01.09.05.51?

johndickinson, 2024-09-06:
Seems OK to me with freetype2 2.13.3-1 on Arch and bambo-studio built from the source master branch. The about window shows it is 1.9.5.51.

However, using nvidia graphics driver seems to break it (No top menu or plate on the prepare and preview tabs). Works fine with vulkan-nouveau 1:24.2.1-1. I will give the nvidia drivers a try in a couple of release versions time and if it is still broken I will create a separate issue.

MackBambu, 2024-09-06:
> However, using nvidia graphics driver seems to break it (No top menu or plate on the prepare and preview tabs). Works fine with vulkan-nouveau 1:24.2.1-1. I will give the nvidia drivers a try in a couple of release versions time and if it is still broken I will create a separate issue.

Thanks for the test. The thread about the gpu driver can be discussed here, I'm not sure yet what the nvidia 560 driver did to cause the whole 3d viewer to not work anymore.
https://github.com/bambulab/BambuStudio/issues/4669

Sanaki, 2024-09-06:
> Does this problem still exist on 01.09.05.51?

Just tested `Bambu_Studio_ubuntu-v01.09.05.51-24.04.AppImage` on Garuda Linux and can confirm it launched fine with no adjustments needed, using freetype2 2.13.3-1 (current). Seems like this issue can safely be closed.