bambulab/BambuStudio #4636: Strength Modifier splits model.

36bananapies, 2024-08-15:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Sonoma 14.5

### Additional system information

2023 Mac Studio

### Printer

X1C

### How to reproduce

Drop part into build area.
Right click part, add modifier (I use cube)
Click Generic Cube Modifier and add more walls (I used 10)
Slice Part - The cube will slice the model along all faces it intersects.

One other thing I found--someone said that this occurs only when they change the part wall number after adding the modifier. For me it does it regardless.
[image]
[image]

### Actual results

[image]
It cuts the model in the slicer and it translates to the print.

### Expected results

What should happen is the modifier should simply add more walls to the area of the generic cube modifier. This is how it works in QIDI Slicer and Orca.

### Project file & Debug log uploads

[S4M_Base_3DPrinted_JV1 v5.3mf.zip](https://github.com/user-attachments/files/16628314/S4M_Base_3DPrinted_JV1.v5.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-08-16:
yep. it does a problem, the modifier splits the wall. it isn't easy to solve. you may need to wait a long time.

Drlambda12, 2024-10-31:
Any news? Orca slicer makes the modifier without problems, should be possible for Bambu Studio, too?

vanowm, 2025-03-02:
> Any news? Orca slicer makes the modifier without problems, should be possible for Bambu Studio, too?

Are you sure? I'm experiencing the same issue in OrcaSlicer v2.3.0beta2