bambulab/BambuStudio #4638: AMS not retaining filament selection either through manual entry or RFID identification

markclt4, 2024-08-15:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS: 13.6.9 (22G830)

### Printer

Bambu Lab A1 Mini

### How to reproduce

RFID identifies filament and then a few seconds later wipes out the identification. When manually entered both through the Bambu Studio Device page and through the panel screen on the printer, the same thing happens.  It accepts the entry and then a few seconds later it clears it out.

### Actual results

[image]

### Expected results

The filament choice should be accepted and retained.

### Project file & Debug log uploads

[AMS Issue.zip](https://github.com/user-attachments/files/16629395/AMS.Issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

markclt4, 2024-08-15:
This is not project dependent, but happens even if no project is loaded.  This is a new issue - I have not been having this problem until today.

markclt4, 2024-08-15:
I just realized I had two instance of Bambu Studio running on two different machines running two different OS's (macOS and WinOS) at the same time.  I think this may have been the cause.  Can you confirm this would cause the issue and if so, this can be closed.

master611, 2024-08-16:
Make sure the firmware of both the printer and AMS is up to date. Outdated firmware can cause compatibility issues. Do application debugging of [3D printer](https://kingroon.com/collections/fdm-3d-printer).

ksermon, 2024-08-25:
I encountered exactly the same issue described. The value saved after closing all app instances and running the identification manually on the printer, but wiped again when the bambu app was opened. The value would not set when retried from the app.

DanBao-Bambu, 2024-09-02:
I did not see the slot reset command sent by Studio in the log you sent. The possible reasons for this issue are as follows:
1. You have opened multiple Studio instances and some instances do not have machine recognized Filament presets, such as PETG-CF. At this point, you need to close multiple instances.
2. The printer has reset the instance on its own. You can try restarting or logging in to your account again to get the latest cloud settings for the printer and solve this problem.
If there is still this issue, please send the printer log for us to analyze in more detail.

In addition, the latest version should have been released last week, which should integrate the latest preset package. Using this version should be able to solve some reset issues caused by the lack of PETG-CF preset locally in Studio.
Can you let me know if there is still this issue after using the latest Studio version?