bambulab/BambuStudio #4640: Option to globally disable AUX fan - which causes adhesion issues

tomasz90, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
AUX fan cause to lift my prints, often ruin them completly. For instance I have naver had issues with PLA printing on different printer. On P1S if AUX is on, PLA is always warping.

**Describe the solution you'd like**
 I would like to be able to diable it globally.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

RicVilla78, 2024-08-16:
you can disable in all you PLA profiles directly under cooling.
[image]

tomasz90, 2024-08-16:
> you can disable in all you PLA profiles directly under cooling. [...]

I use AMS, I would need to make copy of every filament preset and remember to use them... VERY INCONVENIENT.

RicVilla78, 2024-08-16:
I use AMS all the time….you are supposed to have a profile for each filament…if you don’t, you are not using the system as intended. If you only use one profile for all your PLAs then deactivate it there and use that one. I don’t understand what your problem is honestly. Putting a global switch to deactivate it makes no sense and specifically for PLA the aux fan is very handy to print fast, warping is just a symptom of another problem you may have.

tomasz90, 2024-08-16:
From your perspective, I need to create copy of every filament I bought from bambu and overwrite this setting???? 
So what is the point of the NFC stickers????

RicVilla78, 2024-08-16:
Yes and no, the purpose of the RFID is only to know the type, color, nozzle and bed temperatures and how much filament is it left. I doesnt store a full filament profile. Also no if the fan will be on or off, that is stored in the filament profile.

I you have 3 types of PLA, copy those and uncheck the Aux fan box and leave the original profile as "backup" in case something goes wrong. Probably you can put in in some of the G-code but I am not sure in which order does Bambu Studio processes all the different Start G-codes (Printer, Filament, Process...probably in that order).

tomasz90, 2024-08-17:
Not only me sees it as an issue ;/
Yesterday AUX fan broken my 10 hours print, because it turned on while printing PVA interface and ruin PLA adhesion... I had different profile for PLA, but NOT PVA. This is ridicolous to create every different profile for bambu stock filament ;/

[image]

tomasz90, 2024-08-17:
This is joke
[image]

RicVilla78, 2024-08-17:
Although I still don’t think your proposal makes sense, I have something it may help you. I don’t have a p1S so I don’t know if that works, but the x1c it can be deactivated under the maintenance menu. May be in the p1s is also available.

[image]

hippoduck, 2024-08-22:
The aux fan is the bain of my life. I print lots of small items, and they always fail around the aux fan. It's just cooling the part and the plate too much on that side. 

I created filament profiles but it's easy to forget to change it, and wake up to a failed print. 

If it really helps, then why is it only on one side? Surely that means uneven cooling. I'd love a setting to disable it on my P1S, but may just have to pull the fan cable.

tomasz90, 2024-09-04:
Another print ruined by this f*cking fan... I will not wait for bambu fix...
[image]

hippoduck, 2024-09-04:
@tomasz90 Assuming you made that connector. Mine has the greyscale cable all the way to the fan.

lar3ry, 2024-09-13:
I would REALLY like a way to either set a new profile as default or the ability to change a Bambua Lab supplied profile and save i under the same name.

tomasz90, 2024-09-27:
Exactly... Change the default settings under the same name, with some reset button - it would do the trick.

sol3uk, 2024-10-23:
I'm really surprised this isn't a setting. I'm constantly having warping issues with PLA on only the side that the Aux fan is blasting away at the print. When I turn it off, it prints fine. But making overriding profiles for each filament is such a non solution.

tomasz90, 2024-10-23:
@sol3uk just cut off the wire, I doubt that bambu will fix this in the near future..

3dfdm, 2025-05-10:
A simple "disable aux fan" option that overridesall filament profiles wouldbe good.
I have seen a very lot of aux fan issues on multiple social media platforms that would be resolved ifonly we could permanently disable it.
Whilst the aux fan may have it's use (I'm not 100% sure where that is) there is also the need by many users to not have one.

houmark, 2025-05-10:
> A simple "disable aux fan" option that overridesall filament profiles wouldbe good. I have seen a very lot of aux fan issues on multiple social media platforms that would be resolved ifonly we could [...]

Yes a global setting would be very nice. With 3 AMS units and 12 colors loaded, it's pretty time consuming to change all the colors loaded to a different filament profile, where the only change from the original profile is setting the AUX to 0%.

I have run without the AUX fan for months now and was considering to physically disconnect it. Then I printed two different funnels the other day and had really, really bad surfaces both on the outside and the inside on both models which were quite different in design actually.

After considering options and trying a few different test prints I realized that the hot end comes around to the next layer so fast on these types of models that the prior layer has not yet stabilized or cooled down enough. The result is that the layers are bending outward (on the outside and inward on the inside) and makes the surface really bad.

I turned back on the AUX fan to the standard 70% and the funnel printed with near perfect surfaces. Night and day difference.

That made me consider if I should try the AUX fan again on a more constant basis, even though with Sunlu PLA+ (my main filament) I had a lot of warping on models in the past, even using SuperTack plate, which have given me the very best sticking of any plate I used up until now (to the point where some prints can be hard to get off).

Well, the first print I did warped a bit in two of the four corners, so I guess I am back to turning it off 95% of the time and only back on when doing smaller (round) pieces where the toolhead is moving upward in layers very fast.