bambulab/BambuStudio #4641: crash while slicing

unprolix, 2024-08-16:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

OS Name	Microsoft Windows 11 Pro
Version	10.0.22631 Build 22631
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Manufacturer	Micro-Star International Co., Ltd.
System Model	MS-7E07
System Type	x64-based PC
Processor	Intel(R) Core(TM) i9-14900K, 3200 Mhz, 24 Core(s), 32 Logical Processor(s)
Installed Physical Memory (RAM)	96.0 GB
Total Physical Memory	95.8 GB
Available Physical Memory	82.9 GB
Total Virtual Memory	110 GB
Available Virtual Memory	93.2 GB
Kernel DMA Protection	On
Virtualisation-based security	Running
Virtualisation-based security required security properties	
Virtualisation-based security available security properties	Base Virtualisation Support, Secure Boot, DMA Protection, UEFI Code Readonly, SMM Security Mitigations 1.0, Mode Based Execution Control, APIC Virtualisation
Virtualisation-based security services configured	
Virtualisation-based security services running	
Windows Defender Application Control policy	Enforced
Windows Defender Application Control user mode policy	Off
Device Encryption Support	Elevation Required to View

### Printer

X1 Carbon w/AMS

### How to reproduce

Open model.
Slice model.

### Actual results

Crash happens, regardless of settings/parameters

### Expected results

The model should be sliced and ready to print

### Project file & Debug log uploads

[problem.zip](https://github.com/user-attachments/files/16637356/problem.zip)
[log.zip](https://github.com/user-attachments/files/16637387/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ziehmon, 2024-08-18:
Dude, what? :joy: 

> Available Physical Memory 82.9 GB

ziehmon, 2024-08-18:
I tested your project and unfortunately I cannot reproduce the bug. Slicing works fine for me.  Based on the logs and the provided exception, this seems to be related to your machine/OS memory. At least now you know it's not the project.

ttt888zzz, 2024-08-20:
HI，[unprolix](https://github.com/unprolix)：Please try the following steps，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

unprolix, 2024-08-20:
Setting the processor affinity to 0 through 7 inclusive worked. Thanks for the workaround!

closed 2024-08-20 by QingZhangBambu

unprolix, 2024-08-20:
(FWIW, there's no overclocking involved.)